betaflight/betaflight #784: Horizon mode broken with Betaflight (Float) PID in 3.0.0?

sethyx, 2016-07-18:
In Horizon mode the quad just starts oscillating on Pitch / Roll axis rapidly right after takeoff.
Is this a known issue? I haven't used Float before, to be honest, so I don't know when it broke.

It flies well after reverting back to Legacy / Integer PID and using the same values.

[blackbox_log_2016-07-18_121153.TXT](https://github.com/betaflight/betaflight/files/369418/blackbox_log_2016-07-18_121153.TXT)

borisbstyle, 2016-07-18:
My next tests will be the level modes.

Probably due to new scaling where I didnt consider angle modes.

sethyx, 2016-07-18:
Cool, let me know if you need help with testing.

sethyx, 2016-07-19:
Yep, I can confirm that reverting [6582a95](https://github.com/betaflight/betaflight/commit/6582a958aa5c93a84f97e91abfb851df319860d3) fixes Horizon.
I tried to mock around and multiplied 16 by 4.1f in the angleRate calculation, this gave much better results. I just started digging around in BF code and I don't understand everything, so I don't want to submit a PR yet, probably that's not the solution anyway. :)

gaelj, 2016-07-19:
@borisbstyle why all the magic numbers ?

borisbstyle, 2016-07-19:
@gaelj
You mean the PID float scalers? It is actually all to give people familliar tuning range like old legacy pid controllers. Actually the old rewrite scaling.
Also it was easier to integrate into configurator at the time

Fix is fairly easy guys. I didnt take account of level / horizon modes back at the time while converting the pid controller to the actual deg/sec.
The default pids are now probably too strong

borisbstyle, 2016-07-19:
Ok attempt to fix it.
https://github.com/betaflight/betaflight/commit/52e302e51e07935a5384891e3e23151c1445bdc8

Can you try the latest code?

sethyx, 2016-07-20:
Yes, that flies great!
Thanks Boris, I'm closing this issue.

closed 2016-07-20 by sethyx