betaflight/betaflight #791: Shaky/stuttering behaviour betaflight 2.9.0 on naze32

7fpv, 2016-07-19:
Hello all,

I upgraded to betaflight 2.8.1 a few weeks ago but quad didn't fly as expected (crashed like automatically after a few seconds in the air), so downgraded to 2.5.4 which seemed to solve the problem. Then I re-upgraded to 2.9.0 and played a bit with the PID's and got it to fly well.

I recently crashed pretty hard, and there was like one motor that would not spin anymore and kept making this double tone noise (the esc did). So I replaced motor as that one seemed to be the culprit. Since then there has been a permanent stuttering/shaky problem. (see link to video). I first thought that it was one of the props that was slipping, but it wasn't. I now noticed that it's at each time that I give like a throttle burst, that this shaking occurs.

I looked through the issues and tried a few things such as recalibrating the gyros, increasing min_throttle, but none of them seem to solve the problem.

I am using naze32; cobra 2204 2300kV motors with lumenier simonk 20A esc's. 

It has been a few months of one problem after the other, so I hope someone can help me with this issue. 

[https://www.youtube.com/watch?v=-DUEMMxYtGE&feature=youtu.be](url)

7fpv, 2016-07-19:
[No5_19_7_2016_QAV210_naze32_Betaflight_2_9_0_with_airmode_ISSUE_TO_SOLVE.pdf](https://github.com/betaflight/betaflight/files/372026/No5_19_7_2016_QAV210_naze32_Betaflight_2_9_0_with_airmode_ISSUE_TO_SOLVE.pdf)

Here is a dump in pdf file to avoid filling a whole webpage.

KiteAnton, 2016-07-22:
Go over all your wiring and hardware, the problem is most likely there. Swap esc/motors until you find the failing one. Also examine your motors/escs for visual damage or loose/slipping magnets. If you are using Naze32 rev6 then this has a known problematic gyro that might need soft-mounting for some setups (see more regarding this issue [here](https://github.com/betaflight/betaflight/issues/322)). 

You will probably get faster response with these kind of problems in the forums. 

https://github.com/betaflight/betaflight/blob/master/CONTRIBUTING.md

closed 2016-07-22 by KiteAnton

7fpv, 2016-07-22:
Thanks @KiteAnton,

I understand that this was probably not the right place to ask for support as it is probably a hardware issue. I'll keep this in mind for the future! Thanks anyways for the suggestions.