betaflight/betaflight #793: Kiss Pitch and roll stuck after input

QuadroBro, 2016-07-19:
Hi, I just flashed 3.0.0 to the kiss board and noticed that after pitch or roll input, the motors stick there. This is with airmode and superexpo off. Rather than return to throttle value, although stick input returns to center, it is stuck at the last value.

To rule out stupidity on my end, here's a dump:

Entering CLI Mode, type 'exit' to return, or 'help'

# dump all

# version
# BetaFlight/KISSFC 3.0.0 Jul 18 2016 / 19:29:11 (4c3d098)
# name
name -

# dump master

# mixer
mixer CUSTOM
mmix reset
mmix 0 1.000 -1.000 1.000 -1.000
mmix 1 1.000 -1.000 -1.000 1.000
mmix 2 1.000 1.000 -1.000 -1.000
mmix 3 1.000 1.000 1.000 1.000
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature -OSD
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
[... 474 lines cut ...]
# rateprofile
rateprofile 2

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

# restore original rateprofile selection
rateprofile 0

# restore original profile selection
profile 0

# save configuration
save

#

borisbstyle, 2016-07-19:
I see feature airmode is off. That means no zero throttle control when
airmode not activated what happens once you pass 35% of throttle

set zero_throttle_stabilisation = on

That will enable stabilisation at all times

QuadroBro, 2016-07-20:
I don't want airmode on, but I will make that change and see what happens.

borisbstyle, 2016-08-02:
it was configurator issue

closed 2016-08-02 by borisbstyle