betaflight/betaflight #794: new led functions not working correct like cleanflight led's

TheAngularity, 2016-07-19:
- warning overlays not working correct (arming state warning)
- color function is set on all clean led's ass default
- imho the led is very slow, cf runs faster

# led betaflight
led 0 11,8::A:0
led 1 10,8::A:0
led 2 9,8::AW:0
led 3 8,8::AW:0
led 4 7,8::AW:0
led 5 6,8::AW:0
led 6 5,8::A:0
led 7 4,8::A:0
led 8 4,4::C:6
led 9 5,4::CW:6
led 10 6,4::CW:6
led 11 9,4::CW:6
led 12 10,4::CW:6
led 13 11,4::C:6
led 14 0,0::C:0
led 15 0,0::C:0
led 16 0,0::C:0
led 17 0,0::C:0
led 18 0,0::C:0
led 19 0,0::C:0
led 20 0,0::C:0
led 21 0,0::C:0
led 22 0,0::C:0
led 23 0,0::C:0
led 24 0,0::C:0
led 25 0,0::C:0
led 26 0,0::C:0
led 27 0,0::C:0
led 28 0,0::C:0
led 29 0,0::C:0
led 30 0,0::C:0
led 31 0,0::C:0

# led cleanflight
led 0 4,8::A:0
led 1 5,8::A:0
led 2 6,8::WA:0
led 3 7,8::WA:0
led 4 8,8::WA:0
led 5 9,8::WA:0
led 6 10,8::A:0
led 7 11,8::A:0
led 8 11,4::C:6
led 9 10,4::WC:6
led 10 9,4::WC:6
led 11 6,4::WC:6
led 12 5,4::WC:6
led 13 4,4::C:6
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

gaelj, 2016-07-19:
- The warning overlays show arming diabled (blink green), low battery (blink red), and failsafe (yellow / blue)
- Color function is now used as default ; indeed, this is a feature of the new led encoding. You can set color 0 if you need a blank LED.
- Here it runs really fast... There was a separate commit to increase the speed back to normal (which was merged) : https://github.com/betaflight/betaflight/commit/2e1766fbb9df666c37f0660aca6b850dfdca6084 are you sure you have it ?

TheAngularity, 2016-07-20:
> The warning overlays show arming diabled (blink green), low battery (blink red), and failsafe (yellow / blue)

new bench test now (using Matek RGB LED Stripe WS2812B 8 LED on back and Matek RGB LED WS2812B & Buzzer on front)
- arming disabled -> not working (only led on fc)
- low battery -> working
- failsafe -> working

> Color function is now used as default ; indeed, this is a feature of the new led encoding. You can set color 0 if you need a blank LED.

ok, a new feature ... I thought it was a bug

> Here it runs really fast... There was a separate commit to increase the speed back to normal (which was merged) : 2e1766f are you sure you have it ?

yes i have it ... is just maybe my feeling :)

gaelj, 2016-07-20:
Can you for instance tilt your craft to be sure arming is disabled ? It's working on my setup.
Regarding speed, you can try playing with the value in the PR I mentioned. I'll compare with CF this evening.

TheAngularity, 2016-07-20:
Yes, i test it in my hand and tilt the craft to disable the arming.

So i make a new test.
If i set the warning overlay led's with color 0 the arming disabled blink is working

# led
led 0 4,8::A:0
led 1 5,8::A:0
led 2 6,8::CW:0
led 3 7,8::CW:0
led 4 8,8::CW:0
led 5 9,8::CW:0
led 6 10,8::A:0
led 7 11,8::A:0
led 8 4,4::C:6
led 9 5,4::CW:0
led 10 6,4::CW:0
led 11 9,4::CW:0
led 12 10,4::CW:0
led 13 11,4::C:6
led 14 0,0::C:0
led 15 0,0::C:0
led 16 0,0::C:0
led 17 0,0::C:0
led 18 0,0::C:0
led 19 0,0::C:0
led 20 0,0::C:0
led 21 0,0::C:0
led 22 0,0::C:0
led 23 0,0::C:0
led 24 0,0::C:0
led 25 0,0::C:0
led 26 0,0::C:0
led 27 0,0::C:0
led 28 0,0::C:0
led 29 0,0::C:0
led 30 0,0::C:0
led 31 0,0::C:0

If i set the color to 6 (green) the arming disabled blink is not working :(

# led
led 0 4,8::A:0
led 1 5,8::A:0
led 2 6,8::CW:6
led 3 7,8::CW:6
led 4 8,8::CW:6
led 5 9,8::CW:6
led 6 10,8::A:0
led 7 11,8::A:0
led 8 4,4::C:6
led 9 5,4::CW:6
led 10 6,4::CW:6
led 11 9,4::CW:6
led 12 10,4::CW:6
led 13 11,4::C:6
led 14 0,0::C:0
led 15 0,0::C:0
led 16 0,0::C:0
led 17 0,0::C:0
led 18 0,0::C:0
led 19 0,0::C:0
led 20 0,0::C:0
led 21 0,0::C:0
led 22 0,0::C:0
led 23 0,0::C:0
led 24 0,0::C:0
led 25 0,0::C:0
led 26 0,0::C:0
led 27 0,0::C:0
led 28 0,0::C:0
led 29 0,0::C:0
led 30 0,0::C:0
led 31 0,0::C:0

TheAngularity, 2016-07-20:
I have it fixed now in #801

closed 2016-07-20 by TheAngularity