betaflight/betaflight #797: Flight- Bad Drift issues in Angle/Horizon Mode

rgluckman, 2016-07-20:
3 of my Quads drift heavily in Angle/Horizon modes. 
All are using Cyclones.

MotoLab Cyclone FC
BLHeli-14.6
Betaflight 2.9

No matter how many times I calibrate the Accelerometer and do an "in flight calibration" my quads won't hold a level hover. The more I fly the worse it gets. It pulls in all directions. 
I tried silicone dampeners on the FC thinking it was a vibration issue. That didn't work either. 
My TBS Powercube and Naze32 Rev5 hold Angles modes just fine. 
What else can I do to fix this?

borisbstyle, 2016-07-20:
Reduce noise in the system. It has nothing to do with acc calibration most
likely.

MPU6500 is known for vibration sensitivity

rgluckman, 2016-07-20:
How do I reduce noise?

rgluckman, 2016-07-20:
The Cyclone uses a low-noise MPU-6000. I don't believe it has the MPU6500. What can I do to trouble shoot this?
I read somewhere to turn "damped light" off.

ctzsnooze, 2016-07-20:
There was a long discussion about failure to hold level mode angle on the cleanflight boards:
https://github.com/cleanflight/cleanflight/issues/791
At that time the issue was due the way the idea of 'level' is derived.  It is not a calibration issue; the acc knows precisely what 'level' is.  
I can't understand the relevant code in betaflight as it uses a different coding to determine what 'level' actually is while in flight.
The problem is that an accelerometer cannot determine what 'level' (as in 'horizontal') actually means while the quad is moving.  Somehow it has to know it is still.  So only while the gyro is reading basically zero can the accelerometer determine the actual angle of the quad.  Any time the quad is moving, the current angle of the quad has to be determined from the cumulative change in angle over time from the last 'certain' value based on gyro data.  And if gyro and acc data are noisy, this becomes very difficult indeed.  The actual offset you're getting is due in part because of noise affecting the FC's ability to determine true level after a period of noisy data, and also is due to an aliasing artefact that arises when sampling of the accelerometer values is phase associated with that noise (folding down effect from aliasing).
The gyro on the rev5 naze is the 6050, the i2c variant of the 6000 on cyclones.
I have about 15 cyclones now, not all on active quads, and don't use any level modes, but recently I enabled it on some and it seemed fine, so it isn't fundamentally an issue with the board.
To be honest this issue should be left open but it would take a lot of time and testing to work through, and since very few pilots here use level modes I doubt anyone has the time to really look into it.  The cleanflight guys might - lots more level mode users there.

martinbudden, 2016-07-20:
@borisbstyle , this is because the accelerometer has been changed to run every 10ms instead of every 1ms. This causes aliasing which in turn causes drift. This was an issue in Cleanflight a while back and was fixed by changing the accelerometer to update every 1ms.

`taskUpdateAccelerometer` is fairly lightweight and this does not cause a bandwidth problem.

rgluckman, 2016-07-20:
In the morning I'm going to delete Betaflight and try the newest Cleanflight and see what happens there. 
My Naze32 and Colibri Race FC (Powercube) and all my Cyclones all have the exact same settings. All the same 14.6 BLHeli, 2.9 Betaflight, and all settings in the configuration tabs are the same. Both the Powercube and Naze fly great in angle/horizon modes. The Cyclones are the issue. I use Futaba so my Mid RC is 1520. In my radio I set end points to 1000/2000. Will it help if I do a "Dump" so you can see my setup?

rgluckman, 2016-07-20:
Also. After a level calibration and 10 seconds after a simple flight in Angle mode the drift was almost full forward stick.

ctzsnooze, 2016-07-20:
@rgluckman Thanks for offering to send the dump but this is due to aliasing of the acc signal, not a configuration or filtering thing.  It might be interesting to confirm current cleanflight is OK.

Why it happens with the cyclone is interesting.  Cyclone uses spi to talk to the gyro, not i2c like naze.    Maybe that is a contributing factor.  Wait till a patch comes out with revised acc task and see if that fixes it.

borisbstyle, 2016-07-20:
There is no aliasing.

Acc task is always running at 1khz, but gyro task has advantage over acc.
So if you are running high speed looptime 8k etc than acc task might get
skipped in order to give priority to gyro and pid task.

For optimal level modes just try pid loop at 1k and maybe lower gyro speed
as well.

Maybe even good to try gyro_lpf = 188hz as well.
But also softmount your FC

borisbstyle, 2016-07-20:
@martinbudden 
Its actually rescheduled
https://github.com/betaflight/betaflight/blob/master/src/main/main.c#L698-L709

martinbudden, 2016-07-20:
My mistake, I had not seen that.

rgluckman, 2016-07-20:
The FC is mounted to the frame and 8 Silicone washers have been installed. FC is mounted above the PDB. How else can I soft mount it? I will try the Gyro_lpf = 188hz

borisbstyle, 2016-07-20:
Well that sounds fine than

rgluckman, 2016-07-20:
gyro_lpf = 188hz - Didn't make any difference. 
1000 Looptime - Didn't make any difference. 
I also turned off Multishot with no difference. 
Turned Off "Damped Light" and put on "Low"
That made no difference.

rgluckman, 2016-07-20:
Cleanflight 1.13 Firmware at 2000 Looptime made no difference. I think I've about given up. Do you guys know of a FC that works in Angle Modes besides Naze32 Rev5?

borisbstyle, 2016-07-20:
Its your setup rather than fc or firmware.

Try @digitalentity iNav firmware as that one should have the best level /
horizon / gps focus at the moment

rgluckman, 2016-07-20:
Thanks Boris. I'm trashing these 3 Cyclone FC's. 
What other FC is good?

borisbstyle, 2016-07-20:
Actually cyclone is really good, but there are many new boards out there

ctzsnooze, 2016-07-21:
Swap a known good FC board from a quad that 'works' into that quad, load the same settings, and see what happens.  And put the 'bad' cyclone into a 'good' quad.  if that particular cyclone board seems to not work, it could be a hardware issue and one of the other three may work.  See if you can get a blackbox recording.

sethyx, 2016-07-21:
@rgluckman I just rebuilt my 250 with an [XRacer v3](http://www.fpvmodel.com/x-racer-f303-flight-controller-v3-1_g1106.html) (X_RACERSPI), it flies pretty well in level modes too. Only if you don't need a built-in baro.

rgluckman, 2016-07-21:
@sethyx. I'll check that out. Is that an F3?

sethyx, 2016-07-21:
@rgluckman Yes, and w/ MPU6000, which is great IMO. Maiden flight with default PIDs and slightly lowered level PIDs on the latest Betaflight code was almost perfect. I still consider myself as an amateur but I'll try to make a video with blackbox overlay today evening.

andre92, 2016-07-21:
@rgluckman I assume the copter flies well in acro mode with no gyro drift? In that case, try lowering acc_lowpass to 10 to avoid aliasing. Could you make a blackbox log of the accelerometer and gyro traces?
Also, try gyro_lpf = 188, it also filters the accelerometer, that may help.
[image]

rgluckman, 2016-07-21:
@andre92 It flies great in Acro mode. Angle mode holds level for 5-10 seconds. As it drifts in any direction I have to give it opposite elevator/aileron stick to keep it center. As the flight gets longer the drift gets worse and worse almost to the point where I run out of elevator/aileron stick to keep it center and in a hover. 
If I flip it into Acro mode the drift goes to 0 and it flies smooth as butter. 
I have done gyro_lpf = 188hz as @borisbstyle told me in an earlier post with no changes whatsoever.  
I have not done acc_lowpass = 10 yet. I will try that tomorrow. 
The Cyclone FC does not come with a flash chip installed. How can I get a Black Box recording for you?

andre92, 2016-07-21:
I'm sorry, I didn't realise you don't have a flash chip. If you cannot make a backbox log, no problem. Still, try playing around with acc_lpf. I had to reduce it on my copter to get better leveling. (I actually got toilet bowling effect with too high acc_lpf) I feel like the default of 15 is a bit too high. The lower it gets the less noise it lets through, it may lag a bit but should not drift as much.
Ps. if you have a spare uart, you can connect an openlog device, I'm not sure how well that works though, I always use onboard spi flash because it logs at 1khz.

siamect, 2016-07-22:
I'm a complete newbie in the quad flying business but I have some experience with electronics and data acquisition circuitry and I have a hard time believing that an aliasing between the vibrations of the copter and any sensor's  acquisition frequency could result in a drift like the one we are discussing here.
What you get is a frequency component that is related to the difference between the acquisition frequency and the frequencies that comes in from the sensors. These would probably be frequencies related to the motor's etc and those ones are not stable so the drift would be going up and down many times per second, with always changing frequencies. This should not result in anything like the 10s forward/backward drift. (unless there is a flaw in the design resulting in a rectification somewhere, which is highly unlikely)

I think maybe the problem is related to 90 or 180 degrees misalignment between the accelerometers and the gyros. The gyros, correct me if I'm wrong, is most important for the acro-mode stabilization and the response to their signals is fast. The angle mode, I guess, is depending on the accelerometers? and if they are misaligned 180 degrees it will pull the opposite direction to what you want. The gyros however will give signals to fight this and win short term but loose long term and the angel drift will slowly go to the extreme end. 
There is somewhere in the settings where you can set the angle of the flight controller, for example if you have turned it upside down, also another setting that on mine is set to "default" and if I remember correctly, that is the angle of the accelerometers. I would play a little with these if I had this problems.

ctzsnooze, 2016-07-22:
It's easy to check acc alignment.  With motors off the acc works normally.  This is easily confirmed in the GUI.  I've had this problem myself.  It is noise related and due to aliasing, that has been confirmed before.  We don't really know why it causes drift on one angle but it does.  If it is hand-held and slowly throttled up, you'll find that the magnitude and sometimes direction of the drift varies with throttle amount.  Certain frequencies get aliased around the point the quaternion code shifts from acc to gyro as the source of attitude (angle) information.  

andre92's advice above to check the frequency of the acc task is what needs to be done.  And yes lowering the acc_lpf might help.  A black box recording makes a huge difference.

I would also be encouraging trying a second of the three cyclones just in case the acc is faulty on only one of them.  Very improbably it is faulty on all three.

siamect, 2016-07-23:
Yes I have to agree with most of what you say... 
My statement about the alignment is probably not correct. 
But I still fail to see how aliasing can be causing a drift. To make that happen for any longer period of time you need to somehow have a sync between sampling and the vibrations.  
But what could cause this a drift is clipping. Say that you have a signal that represents the acceleration and on top of that you have a sinus shaped vibration. If the vibration is reasonably small we can filter it out by integrating over a period of time (LP filter). But what if the vibration is large enough to cause clipping. The sinus shape is no longer a perfect but it has it's top and bottom chopped off. It would mean two things. 
Any valid signal (what is left after filtering) would be damped as the LP filter would miss the minima and maxima of the signal. 
Secondly and maybe more importantly, clipping would not be perfectly symmetrical. It means that the chopping of the top and the chopping of the bottom would not be the same (because all systems has an offset) and that may be causing a drift of the signal as it comes out after the LP filter. 
We can complicate the example further by assuming a vibration that is not symmetrical. We may in such cases have significant clipping on only one direction and not the other... 
On top of that we have to bring the saturated circuitry back into it's linear range, after the clipping, which will take some time and that time depends on a number of factors so there you certainly have asymmetrical behaviors.
Something like that can explain most of the symptoms described by the OP. 
It could probably be easily verified or dismissed by logging raw samples from the sensors of an affected system.

ctzsnooze, 2016-07-23:
I think that the quaternion code has some method for determining when to use gyro and when to use acc.  It depends on the acc value.  If the noise causes acc to transition through that point in a repeatable manner, and if the noise has a vector at that time, it could cause offsets because both the gyro value and the acc value at any point in time are related but filtered differently, and the noise causing the problem can definitely have a specific vector that then gets trapped.

I was able to greatly improve this behaviour when it happened to me under the old 'complementary filter' (since replaced by the quaternion code) by changing crossover related filter values.  I do not know how the quaternion code operates but suspect it is at the root of it.

This problem affects small numbers of users and has always been a bit mysterious.  Affecting only a small number of users sometimes implied a gyro fault or a particularly or peculiarly noisy quad. Low pass filtering acc is good in theory but much lower than 10 impairs responsiveness and leads to wobble.

I think it's important that the OP confirm the issue happens on two different cyclone boards, see if it is throttle related, and if possible gets a log.  Drift is very difficult to see in a log since it is such a small value but the noisiness of the particular system isn't so difficult to see.

siamect, 2016-07-27:
Thanks CTXsnooze
I think I see it clearer now and after reading through the cleanflight issue 791 carefully it makes more sense. 
So to determine orientation, the accelerometers are used but only when they are close to 1G etc and in other cases only the gyros are used to determine orientation but as the gyros are drifting this works only short term. Due to  vibrations the accelerometers can be outside the range where they are used for orientation more or less all the time and only the gyros are used. In that situation there is no way to determine/compensate for the gyro drift. 
If this is correct, it makes more sense to me...

borisbstyle, 2016-07-27:
Closing this

closed 2016-07-27 by borisbstyle

modellbobby, 2017-10-01:
quick check here. Why is this closed. It is still present in 3.2 RC5 badly

VoicOfReason, 2017-10-02:
I agree. I am having this issue, as well. I'm using a Talon frame. I had the issue on Betaflight F3 board with hyperlite 2007 motors. I changed the motors (to 2307) and the FC to F7V2 and I have the same issue. It rises perfectly level and then after 1-2 throttle punches, it drifts HARD to the right, so hard sometimes that I almost crash since I have to crank the stick all the way just to keep it level.

modellbobby, 2017-10-02:
Hi, this bug is referring to : Acc filtering is broken. #4226

modellbobby, 2017-10-02:
Duplicate of #4226

modellbobby, 2017-10-02:
#4226