betaflight/betaflight #799: KISS FC - Motors 3 and 4 don't work

kitsen13, 2016-07-20:
Hello,
When I tried Betaflight DEV 3.0.0 on Kiss, motors 3 and 4 don't spin (3,4 based on kiss mapping).
Tests was made on "motor test page" of "Betaflight configurator" individualy motor by motor.
Strange thing, when I test with the master cursor, motor 3 and 4 begin to spin at 50% of command.
I cannot try with with my D4R receiver because PPM doesn't work too.

kitsen13, 2016-07-20:
[image]

borisbstyle, 2016-07-20:
Try unsynced fast pwm.

Some reported that it works with that

borisbstyle, 2016-07-20:
Can you perpaps test PPM on AUX2?

QuadroBro, 2016-07-20:
Boris is right. I think I posted that in the RC groups thread too. Unsynced oneshot is the only way I got all 4 to spin. You are probably gonna need to remap as well

llambkin, 2016-07-20:
Shouldnt need a remap, we solved that one about 3 weeks ago :D

borisbstyle, 2016-07-20:
I think I know why the issue is there.

Kiss uses this timer config for motor 3 and 4.
TIM1_BRK_TIM15_IRQn

During sync pwm operation timers need to be stopped each time after first
pulse. I guess the above timer cannot be stopped

kitsen13, 2016-07-20:
Test made : unsynced activated.

Motor 3 and 4 work, but now it's motor 1 and 2 what don't work..

llambkin, 2016-07-23:
I can confirm that there are issues with the motor output, I can only run unsynced to get it working. I cant remember weather it was 1khz unsynced or 2khz that got it working, but definitely a bug there.....

This was on my second quad with XM20A ESC's, not that the ESC's should have anything to do with it....

QuadroBro, 2016-07-27:
@llambkin Are you sure? I had to remap 3 and 4. I can only run unsynced and only running os125. Would love to be able to flash MS back on the esc's though. Would that be possible in a future release?

QuadroBro, 2016-07-27:
@kitsen13 Make sure your esc's are wired for the Kiss motor layout and that they are running standard blheli. In the configurator you need to set the motors to unsynced and oneshot 125. Lets make this easy. Please post or link screens of your blheli configuration, as well as the configuration page from betaflight configurator.

Last note: Currently the best looptimes I've been able to run stable are 4k/2k and the motors should be unsynced at 2k or lower. They either won't spin or desync any higher.

borisbstyle, 2016-07-27:
Someone else tested this for me and unsynced option always works.
It is just synced what causes trouble

borisbstyle, 2016-07-28:
@martinbudden 
Can you once more go through the synced vs unsynced code to see why it would fail on this target.

KISS is the only one with inverted motor output. There is a possibility that inversion is canceled or not applied in synced mode. You know the oneshot override code etc. I couldnt see the issue

martinbudden, 2016-07-29:
@borisbstyle ,
I've had a brief look at the code and cannot see anything obvious. I'll have a further look. I am working in the dark though, no KISS FC or schematics :-) I did email Flyduino to ask if they would send me one, but I didn't even get a reply.

Anyway, I noticed the `target.c` file has changed:
```
    { TIM1,  IO_TAG(PB14), TIM_Channel_2, TIM1_CC_IRQn,            TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_N_CHANNEL, IOCFG_AF_PP, GPIO_AF_6}, // TIM1_CH2N
    { TIM1,  IO_TAG(PB15), TIM_Channel_3, TIM1_CC_IRQn,            TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_N_CHANNEL, IOCFG_AF_PP, GPIO_AF_4}, // TIM1_CH3N
    //{ TIM15, IO_TAG(PB14), TIM_Channel_1, TIM1_BRK_TIM15_IRQn,     TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_INVERTED, IOCFG_AF_PP, GPIO_AF_1},
    //{ TIM15, IO_TAG(PB15), TIM_Channel_2, TIM1_BRK_TIM15_IRQn,     TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_INVERTED, IOCFG_AF_PP, GPIO_AF_1},
```

Can you recall why that change was made?

Also, a shot in the dark here, would it be worth trying?
```
    { TIM1,  IO_TAG(PB14), TIM_Channel_2, TIM1_CC_IRQn,            TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_INVERTED|TIMER_OUTPUT_N_CHANNEL, IOCFG_AF_PP, GPIO_AF_6},
    { TIM1,  IO_TAG(PB15), TIM_Channel_3, TIM1_CC_IRQn,            TIMER_OUTPUT_ENABLED|TIMER_OUTPUT_INVERTED|TIMER_OUTPUT_N_CHANNEL, IOCFG_AF_PP, GPIO_AF_4},
```

borisbstyle, 2016-07-29:
@martinbudden 
That was another shot in the dark by @nathantsoi :D. Just some trial and error. He now has the board by the way so he can test it.

Can you motivate why you think your change will fix it? Is it something another targets uses with success?

martinbudden, 2016-07-29:
No real reason why I thought it might fix it. Thought would be worth a try, since I can't see anything else. @nathantsoi could you try this if you have time.

I'll spend some more time looking through the code later today.

borisbstyle, 2016-07-29:
Check chapter 3.14 for timers
http://www.st.com/content/ccc/resource/technical/document/datasheet/d1/cd/3d/18/a2/2c/4e/d0/DM00097745.pdf/files/DM00097745.pdf/jcr:content/translations/en.DM00097745.pdf

borisbstyle, 2016-07-29:
Lets try out your change. Make a PR so nathanosoi can try

borisbstyle, 2016-07-29:
NEver mind I did it. I just pushed it. It seems right from the datasheet.
https://github.com/betaflight/betaflight/commit/b6c6475d4cb7a8e3d7ab8b168df942e758afc1e4

llambkin, 2016-07-30:
I will give it a try when I get home in a few hours time. I still have a KISS FC on another craft....

nathantsoi, 2016-07-30:
i think this is fixed by https://github.com/betaflight/betaflight/pull/849, could you test and confirm @llambkin ?

llambkin, 2016-07-30:
Confirmed that the unsynced problem is fixed, but now the PID controller doesnt work. The motors spin up to both throttle input and motors tab, but no reactions to sticks or moving the craft whilst armed.

I have tried with both synced and unsynced PWM and still no reactions.

llambkin, 2016-07-30:
But in saying that, I have removed my buffers on my board, so I can use BLheli passthrough support. So my board isnt exactly standard.

nathantsoi, 2016-07-30:
ru using ppm input or serial rx?

llambkin, 2016-07-30:
Serial RX, I am on gitter right now

martinbudden, 2016-07-30:
@llambkin , don't you need to do a custom build, removing the `TIMER_OUTPUT_INVERTED` flags from `target.c` ?

llambkin, 2016-07-30:
Yes, I have been doing so. Then doing make and flashing the output hex.

martinbudden, 2016-07-30:
OK, I thought I'd just check. It's easy to sometimes forget to do these things...

nathantsoi, 2016-07-30:
thx all. motors seem fixed. closing this issue.

closed 2016-07-30 by nathantsoi