betaflight/betaflight #804: Support for SPI Micro SD Card reader on targets with SPI pads

llambkin, 2016-07-21:
Hey Guys,

I was wondering if it is possible to implement an SPI card reader like this one:
http://www.ebay.com.au/itm/like/331911162601?lpid=107&chn=ps
And weather this might have better throughput than current serial logging. I have a CC3D Revo that doesnt have its OPlink board attached, and I think it attaches using SPI.

So since it is not there, are we able to attach a SPI datalogger and have it work? Or is this too involved?

-Liam

MJ666, 2016-07-21:
This should definitely work. This reader is for 5v power. A more simple one for 3.3V only would be sufficient. You need to make an custom target to include the SDcard support. Will not be very complicated.

llambkin, 2016-07-21:
Would I just simply copy some of the code from the SPRF3EVO target.h file? Obviously I'd have to make note of where the pads actually go to on the REVO's CPU, but is it that simple? Just tell the target.h that these pads are SPI SD card, and that is it?

MJ666, 2016-07-21:
You need to activate the proper SPI port and also add the SDcard related settings and update them in the target.h file. Best get the settings from on target wich is using the same SPI port. In the target.mk file you need to add the SDCARD feature.

blckmn, 2016-07-25:
I think we can do this in a future revision when we have parameter groups, and the configuration of a target beyond defaults by pin allocation.

DeepnineRC, 2016-07-25:
[image]

I'm doing this to all my flight controllers since sd card code is available. Just replace flash memory with sd card. Better solution is a micro sd adapter, can be solder better. is cheap.

update target.mk
SDCARD
update target.h:
#define USE_SPI
#define USE_SPI_DEVICE_2 // PB12,13,14,15 on AF5

#define SPI2_NSS_PIN            PB12
#define SPI2_SCK_PIN            PB13
#define SPI2_MISO_PIN           PB14
#define SPI2_MOSI_PIN           PB15

#define USE_SDCARD
#define USE_SDCARD_SPI2

#define SDCARD_DETECT_INVERTED

#define SDCARD_SPI_INSTANCE                 SPI2
#define SDCARD_SPI_CS_PIN                   SPI2_NSS_PIN

#define SDCARD_SPI_INITIALIZATION_CLOCK_DIVIDER 128
#define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER     2
#define SDCARD_DMA_CHANNEL_TX               DMA1_Channel5
#define SDCARD_DMA_CHANNEL_TX_COMPLETE_FLAG DMA1_FLAG_TC5

astragalizont, 2016-09-02:
Hi DeepnineRC!

Please contact me via kopterforum.de

thanks!

Extent421, 2016-12-03:
This is even easier on the mini revo where they break the "oplink air" port out to a connector instead of just leaving the RF module unpopulated.  I printed a custom board with reg and decoupling.  
[image]
My target is here
https://github.com/Extent421/betaflight/tree/miniRevo

llambkin, 2016-12-03:
That is just cool :D

atomiclama, 2016-12-20:
@Extent421 Care to share the pcb details, so others may get some built up?

closed 2016-12-20 by borisbstyle

Extent421, 2016-12-24:
There really are no details, it's literally just an sdcard socket with a regulator and 2 decoupling caps

mikeller, 2016-12-26:
@Extent421 would you be willing to share the PCB design?

astragalizont, 2017-02-03:
Would that work on an F4 Targets too?