betaflight/betaflight #809: Unify pro smartaudio support

jaimevisser, 2016-07-22:
I have a spare UART and would like to setup my Unify Pro HV using my FC. SmartAudio is a UART port so it should be possible to use it like the Core OSD would. Would be awesome to use stick commands+aux switch to setup my VTX without having to get out of my comfy chair. 

Bonus: Output current band/channel/mw using SmartPort telemetry if at all possible to see the current settings on my Taranis.

See also:
https://github.com/cleanflight/cleanflight/issues/2281
d-ronin/dRonin#1067

ps915, 2016-07-22:
That would be awesome.

DaTwo, 2016-07-26:
This would be great!

MattWilliams1, 2016-07-31:
Yes please!

nexudx, 2016-08-03:
yes!! would be awesome - do it!!

SoulPictures, 2016-08-10:
it seems to be implemented there: https://github.com/d-ronin/dRonin/tree/next/flight/Modules/VTXConfig

rdarby777, 2016-08-17:
Is anyone currently working on this feature?

brycedjohnson, 2016-09-12:
Looks like a spec might be available here: 

> SmartAudio
> SmartAudio is a protocol developed by TBS for OSD / VTx communication. All newer generation OSDs at TBS, and all UNIFY PRO series VTX support SmartAudio V1.0, with the UNIFY PRO HV line we have launched SmartAudio V2.0. The new version supports control for the new features (e.g. PitMode). SmartAudio is a single-wire UART protocol, running over the Audio wire. If you are a OSD or VTx developer interested in adding support for SmartAudio, please create a support ticket with us so we can supply you with the technical documentation:
> ● http://team-blacksheep.freshdesk.com/

Gamma-Software, 2016-09-19:
Here is a TBS smartaudio manual:
[TBS.SmartAudio.Rev07.pdf](https://github.com/betaflight/betaflight/files/479793/TBS.SmartAudio.Rev07.pdf)

aee74, 2016-09-20:
I'd love ti see this on betaflight

jflyper, 2016-10-07:
@jaimevisser 
This one can be closed.
Discussion is #1029 
First cut PR is #1275

jaimevisser, 2016-10-07:
As mentioned by @jflyper this issue is thoroughly duplicated, closing.

closed 2016-10-07 by jaimevisser