betaflight/betaflight #811: DTFC DOGE Sbus + telemetry not work.

Boappp, 2016-07-22:
HI, 
You can not activate the S bus and the same time on the telemetry Uart3, yet it is the only way.

borisbstyle, 2016-07-22:
Thats not possible unfortunatelly. Use other port for telemetry

mikeller, 2016-07-22:
Hi @Boappp. This is possible with my patches that are in 3.0.  But it there are some restrictions:

- only FrSky and LTM telemetry protocols are supported
- the serial port speed for telemetry will be set to the same speed as that the serial Rx uses. Your receiver needs to be capable of supporting this (only openLRSng confirmed working so far).

Let me know how you are going.

borisbstyle, 2016-07-22:
Would softserial hack work? 2 softserials combined for smartport. Thats how you do it on the NAZE :D

Boappp, 2016-07-23:
Ok Thanks you, I juste change a S-port on UART2 :)
This FC is perfect, except the buzzer pad .........

closed 2016-07-26 by borisbstyle