betaflight/betaflight #812: PPM RX support is broken for Alienflight F3 target after V2.9.0

MJ666, 2016-07-22:
PPM receivers don't work with V3.0.0 builds anymore. It still works up to version 2.9.0.

tannewt, 2016-07-22:
How is it broken? My last flight on BF3, my F4, and PPM I was having drop outs that weren't triggering failsafe. I haven't debugged the cause yet.

MJ666, 2016-07-22:
There is no input signal detected at all. Likely you problem is someting else but could be related to the timing. The F4 is using an different timer divider than the F1 and the F3. In my knowledge the code is not taking this into account yet. Also an timer clash with motor output chanels at the same timer is not handeled properly. We now have multiple dividers for the different motor output protocols. Actualy only Sparky and AlienFlight F3 is even taking the timer clash into account.

tannewt, 2016-07-23:
What do you mean timer clash? Mine should be on a separate timer thats not
share with anything else. It works most of the time but everything drops to
zero sporadically. It worked before for me though so it may be hardware
damage from a crash.

MJ666, 2016-07-23:
Timers always have 4 channels. They share some prescaler settings. If this channels are used by motors an the PWM input in the same timer you have the clash wich need to be handeled.

nathantsoi, 2016-07-23:
e.g.

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_mapping.c#L292-L296

and 

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_rx.c#L384-L389

MJ666, 2016-07-23:
The clash function is not taking the changed prescaler factors (oneshot, multishot, etc.) into acount yet. Here it is how it looks for RaceFlight: https://github.com/rs2k/raceflight/blob/raceflight/src/main/drivers/pwm_rx.c#L329-348

tannewt, 2016-07-23:
Are prescalers used across timers or only timer channels?

MJ666, 2016-07-23:
Prescaler is the same for all timer channels in one timer. For the F4 they even get feeder by an different clock. Some timers only run with hlf the clock of the others.

tannewt, 2016-07-23:
K, I'll have to look in the next week or so. I know I'm running it on a
separate timer but its an F4 so the feeder may be different.

MJ666, 2016-07-27:
After some detailed checking of all targets only the AlienFlight F3, Sparky and the CC3D are affected by an timer clash. Still trying to find the root of this bug without much success. Likely need to debug this when im back home. PPM RX are not so usual anymore.

borisbstyle, 2016-07-28:
@MJ666 
Were you able to test this again?

MJ666, 2016-07-28:
Just waiting for feedback from my PPM tester. Serial is still working as expected. Flight tested today with latest dev code.

MJ666, 2016-07-28:
A minute ago I got my feedback and it still does not work. BTW the serial cancel for AlienFlight F3 and Sparky is also 2 (third cannel with VCP included). I already fixed this in the version I send to my tester I also put an fix in for the timer clash. But even in PWM motor mode when the fix is not active there is no PPM input received. BF 2.9.0 is working with the same setup.

borisbstyle, 2016-07-28:
@MJ666 
Maybe we forgot to fix things in pwm_output.c. I believe UART3 pins are being skipped there.

I actually think that this never worked for SPARKY and ALIENWIIF3. That part of code has never been changed tbh. Also UART3 was always being activated with or without PPM

borisbstyle, 2016-07-28:
@MJ666 
See this
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_mapping.c#L131-L134

UART3 ports are being skipped

borisbstyle, 2016-07-28:
@MJ666 
Can you try this PR 
https://github.com/betaflight/betaflight/pull/845

MJ666, 2016-07-28:
Will look into this again. PPM Rx input is definitely working till BetaFlight V2.9.0 and also in CleanFlight. Its may be related to the IO changes. the port initialization is may be done differently now. 

BTW: The pin PA3 is used for the serial RX (UART2 RX) and PPM input (Timer2/Cannel4).

borisbstyle, 2016-07-29:
@MJ666 
The IO changes were only done in 3.0 :D

blckmn, 2016-07-29:
I have a Spektrum - with an Orange RX so can test the PPM on the targets I have (e.g. STM32F3DISCO, SPRACINGF3 clone, BJF4, Naze r5, Naze r6 and now Revo).

blckmn, 2016-07-29:
I also think there is a much better way of doing all this initialisation :) but I'll start that once we have the Dev branch for v3.1 and v3.0 is out.

MJ666, 2016-07-29:
@borisbstyle  V3.0.0 is exactly the version which is affected. Very likely I need to make an hardware debuging session to find this problem when back home. I think I have an PPM RX laying around somewhere.

borisbstyle, 2016-07-29:
@MJ666 Earlier you said it was 2.9?
Honestly I can't imagine how this would have worked in previous version. Must be just coincedence perhaps? That initialisation is everything but clean if you ask me. I agree with @blckmn that it needs cleanup.

@MJ666 
Well 3.0 is about to get releases soon. Do you want this to be fixed in a patch or do you think you will have some time to debug it?

MJ666, 2016-07-29:
I have the feeling the timer interupt does not fire for some reason and we always looking at the wrong spot. But this is only guessing and need to be debuged. Upgraded my development environment to Eclipse Neon and the latetest GNU toolchain some days ago. Need to see how debuging goes when im back home. 

I think this is an minor issue. Most users are using Serial RX anyhow. We should not delay the V3.0.0 release for this. We can fix this independetly. Users can go for 2.9.0 if the realy need PPM RX.

MJ666, 2016-07-30:
Have now setup my test environment with the F3 development target and an PPM RX. I can confirm the findings from my tester 100%. V2.9.0 is working but not V3.0.0. need to start debugging it now.

MJ666, 2016-07-30:
The serial port is not touched when PPM is configured. Only the timer overflow interrupt is firing fro some reason. Actually testing with PWM motor output to avoid problems with the timer clash.

MJ666, 2016-08-02:
For now I have no idea what's going on. From CLI resources output I can see the PA3 pin is owned by PPM IN. The PPM initialization code is also properly executed but the timer is only producing overflow interrupts. As the next step I will disable UART2 completly (remove from target.h) and see if this makes an difference. All does not make much sense to me yet.

borisbstyle, 2016-08-02:
We need to understand this issue. Why is it broken, while in fact nothing
has been changed.

MJ666, 2016-08-02:
Found the problem. Please see PR #872. The last commit of this PR is fixing the initial bug. was changed during the IO updates?

MJ666, 2016-08-03:
Close issue. The related fix is merged.

closed 2016-08-03 by MJ666