betaflight/betaflight #817: SP Evo magnetometer not detected

matphillips, 2016-07-23:
Betaflight 2.9 seems to think there is no magnetometer on the SPRacing Evo. 
Not a hardware issue, the mag is detected by switching over to cleanflight.

RipperDrone, 2016-07-23:
BF is sometimes sluggish to recognize the magneto, even for my SPR3. I need to unplug USB, re-plug in, connect - bingo!

matphillips, 2016-07-23:
Interesting. I played with that a bit, two different boards and trying them powered by a limo while replugging, and not powered too. No affect.

Cleanflight CLI shows MAG=AK8963 in status, so searching for that did find issue #135 which is closed, and more accurately #643 also closed.Before opening this issue, I read through the list of open issues and didn't find one about the mag not working.

In #643 Arakon says "It's disabled because it simply doesn't work right" but it is not clear if this refers to the hardware or the betaflight code. The mag seems to work fine in cleanflight (although I've only just finished the first naze32-ectomy and haven't tested beyond a hover)

Could we leave this issue open until the AK8963 driver is functional?

MJ666, 2016-07-24:
The mage is not yet enabled in the SPRACING EVo target.

MJ666, 2016-07-24:
The mag is still not enabled for the SPRACING EVO target. The driver should work. I have tested this at an Alienflight F3 prototype with an MPU9250 during my last updates.

matphillips, 2016-07-24:
Oddly, while the mag icon in the GUI is not lit, and the CLI reports no mag, the telemetry being sent to the Taranis is showing a heading value that changes correctly as the craft moves. I thought heading was from the compass so it may be a case that it is 'partially' working? Either that or the data is being created from the accelerometer.

daPhoosa, 2016-07-25:
OK... so it's easy to see why this doesn't function in the SP Evo.  The target.h files has the defines related to the Mag commented out:

```
//#define MAG
//#define USE_MPU9250_MAG // Enables bypass configuration
//#define USE_MAG_AK8963
//#define USE_MAG_HMC5883 // External

//#define MAG_AK8963_ALIGN CW90_DEG_FLIP
```
These are enabled in Cleanflight.  Eventually I intend to setup an environment to compile and then I would be able to see what  breaks when these are enabled.  Unfortunately,  that won't be tonight.

matphillips, 2016-07-25:
Well, that clears that up then! Both the reason it's off and its use. Thanks for the detail @daPhoosa .

MJ666, 2016-07-25:
Hi, can you give this an try. Remove the JPG extension before flashing. If this works I can make the related PR.
[image]

gaelj, 2016-07-25:
I'm still curious to know why it was seen as broken. Maybe there was a good reason for that  (or maybe not)

MJ666, 2016-07-25:
The driver (SPI part) was broken a while ago but I fixed it and tested for the AlienFlight F3. For F4 targets (SPI mode) it's still not working.

daPhoosa, 2016-07-25:
I can test that hex tonight, but it will be a little late (2300CST) as have some other things going on earlier.

matphillips, 2016-07-25:
@MJ666 that test hex seems to work just fine.

# status
System Uptime: 12 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:8%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=AK8963
Cycle Time: 507, I2C Errors: 1, config size: 1468
SD card: Manufacturer 0x22, 3917824kB, 05/2016, v0.1, 'MSD20'
Filesystem: Ready
# version
# BetaFlight/SPRACINGF3EVO 3.0.0 Jul 25 2016 / 10:23:55 (750fb11)
#

MJ666, 2016-07-25:
@matphillips Please can you also send the output of the task CLI command. Did you also verify the proper function at the sensor page. Will make the PR if everithing works well. Thanks for testing.

matphillips, 2016-07-25:
Sure thing, task output below.
Calibrating mag on Setup tab seems to go fine. Placing a magnet by the board makes the onscreen model spin, placing it at the other side of the board makes it spin the other way.

The values in the sensor tab seem strange though, X: 0.15 Y: -0.02 Z: 0.24
This is on a test board on the bench, nothing else connected to it. It's also not powered by a lipo and I think that may have an affect actually. Cant power it right now to check but will soon.

```
# tasks
Task list:
00 - (SYSTEM) max: 4us, avg: 0us, rate: 9hz, total: 4ms
01 - (PID) max: 2621us, avg: 65us, rate: 980hz, total: 56sec
 - - (GYRO)  rate: 1960hz
02 - (ACCEL) max: 42741us, avg: 21us, rate: 999hz, total: 8sec
03 - (ATTITUDE) max: 83us, avg: 51us, rate: 99hz, total: 2sec
04 - (RX) max: 112us, avg: 79us, rate: 49hz, total: 1sec
05 - (SERIAL) max: 558us, avg: 2us, rate: 99hz, total: 1sec
06 - (BATTERY) max: 15us, avg: 0us, rate: 49hz, total: 103ms
07 - (BEEPER) max: 10us, avg: 0us, rate: 99hz, total: 60ms
09 - (COMPASS) max: 42698us, avg: 16us, rate: 39hz, total: 412ms
10 - (BARO) max: 123us, avg: 60us, rate: 7518hz, total: 2sec
11 - (ALTITUDE) max: 32us, avg: 1us, rate: 39hz, total: 332ms
13 - (TELEMETRY) max: 13us, avg: 0us, rate: 493hz, total: 1sec
15 - (TRANSPONDER) max: 8us, avg: 0us, rate: 235hz, total: 265ms
```

MJ666, 2016-07-25:
Power by usb or battery does not make a difference. The long maximum time for the compass task also looks a bit strange. Have made the PR #822 to allow more testing.

daPhoosa, 2016-07-26:
Although it appears this fix has already been merged, I just tested it.

```
# status
System Uptime: 228 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:7%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=AK8963
Cycle Time: 506, I2C Errors: 1, config size: 1468
SD card: Manufacturer 0x27, 7782400kB, 05/2011, v3.0, 'SD8GB'
Filesystem: Ready
# tasks
Task list:
00 - (SYSTEM) max: 4us, avg: 0us, rate: 9hz, total: 3ms
01 - (PID) max: 2621us, avg: 63us, rate: 992hz, total: 35sec
 - - (GYRO)  rate: 1984hz
02 - (ACCEL) max: 41us, avg: 1us, rate: 999hz, total: 5sec
03 - (ATTITUDE) max: 82us, avg: 49us, rate: 99hz, total: 1sec
04 - (RX) max: 126us, avg: 78us, rate: 49hz, total: 1sec
05 - (SERIAL) max: 3150us, avg: 2us, rate: 99hz, total: 331ms
06 - (BATTERY) max: 16us, avg: 0us, rate: 49hz, total: 65ms
07 - (BEEPER) max: 10us, avg: 0us, rate: 99hz, total: 38ms
09 - (COMPASS) max: 27us, avg: 0us, rate: 39hz, total: 205ms
10 - (BARO) max: 123us, avg: 60us, rate: 5882hz, total: 1sec
11 - (ALTITUDE) max: 41us, avg: 1us, rate: 39hz, total: 215ms
13 - (TELEMETRY) max: 11us, avg: 0us, rate: 493hz, total: 592ms
15 - (TRANSPONDER) max: 8us, avg: 0us, rate: 249hz, total: 170ms
```

An average compass time of 0us seems odd, but the sensor page showed reasonable Mag output when moving the board.

I also don't understand the Baro rate of 5882Hz.  That seems odd.

MJ666, 2016-07-27:
Update is merged. If there are problems we can open an new issue. Close this issue.

closed 2016-07-27 by MJ666

richard-scott, 2017-05-28:
I've just compiled 3.2.0 #dafac79 and this is still an issue for me. The Magnetometer disables when you activate SBUS for my Spektrum Satellite receiver on UART3.

Do you want a new Issue for that?

richard-scott, 2017-05-28:
actually its behaving really oddly on my kit. Sometimes its enabled and sometimes its not. If I can narrow down the reason why I'll open a new ticket.

richard-scott, 2017-05-28:
ah, it seems that if I have my TX switched on the Mag is not detected.

mikeller, 2017-05-29:
Could it be a case of the RF emissions from your TX causing the mag to be overloaded when it is initialised? Have you tried if it works when you remove the TX for 5 meters or so from the craft?

richard-scott, 2017-05-30:
Nah, the Mag was detected initially by the FC with the TX on the same table. Oooh, I've just tried to reproduce the problem so I could document a new issue report and with the latest release 3.2.0 #323d0ae I can't reproduce it. Maybe things are looking up :-)

richard-scott, 2017-05-30:
ok, it seems to be that the soft reboots (when clicking save & reboot) the mag vanishes in the GUI... unplug the USB and re-connect and its there again. However, the output from "status "in the CLI shows that the Mag hardware is not known about. Odd.