betaflight/betaflight #844: Request for target : Pixracer F427VIT6

Maxzor, 2016-07-28:
Hello, I do not know where to start.

What is needed to run betaflight on this board? I push it cuz i have it obv, but more important because it could serve as a great hybrid platform between APM/PX4 stacks and true racing stacks among which BetaFlight.

Are HAL and F4 support in 3.0 release? Or do you guys use STM dev tools like mbed/cube?
Can provide specs, don't know which are needed tho.
http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f4-series/stm32f427-437/stm32f427vi.html

BR Max

blckmn, 2016-07-28:
The schematic etc is available https://github.com/AUAV-OpenSource/FMUv4-PixRacer/blob/master/R12%20Initial%20Production/PIXRACER-R12.pdf

So it's possible. The tricky bit is it uses a STM32F427. It'd need some serious debugging as it runs at 180mhz, and might have some slight peripheral differences.

Linjieqiang, 2016-07-28:
Amazing.

martinbudden, 2016-07-28:
I think @digitalentity is planning to do this for iNav, so it would be worth coordinating the work with him.

Maxzor, 2016-07-28:
@blckmn I am willing to help debugging as much as I can, but I am really bad at coding.
Does this port need ASM programming, or do you rely on STM tools? Trying to contact @digitalentity

Maxzor, 2016-07-28:
Why on earth would you make two chips running at 168 and 180 MHz damn

Maxzor, 2016-07-28:
[image]
Basically, after the tremendous work by @blckmn done for 3.0, I have this list of files using DEF STM32F40_, is that all to be tweaked on top of target?

tannewt, 2016-07-28:
You should be able to do it all with the files in the target directory.

Maxzor, 2016-07-28:
Yes quite, all the lib files are generic and already have 427 actually. Only the src folder might need updating along with target directory^^

Maxzor, 2016-07-28:
I kinda need to write the flow of my understanding somewhere, pls bear with me...
RC is currently capped at f411, although libs cover full f4 scale.
f427 needs special attention to 180MHz, 2048 flash and likely other features.
Various files like makefile, config, platform, main, pwm_output and timer might need editing. (+target)

blckmn, 2016-07-28:
Yes. I'll prepare a basic target directory and get a pull request going. We will also need a startup file and linker script.

blckmn, 2016-07-28:
@digitalentity have you already started anything? Let's coordinate.

Maxzor, 2016-07-28:
Lol @Linjieqiang you were already playing with stm32f4 AQ32 4 years ago.
Anyway :)

Linjieqiang, 2016-07-29:
I just think pixracer is too expensive to obtain and f405 is enough to use.

Maxzor, 2016-07-29:
This may be more confusing than helping... but trying to push the matter, I read what I could and reported on the header file
Legend grey-stock from Bluejay file, dark grey-modified but not sure, green-modified and pretty sure
[TargetPXR4_h.docx](https://github.com/betaflight/betaflight/files/390395/TargetPXR4_h.docx)

digitalentity, 2016-07-29:
@blckmn hi, sorry for being silent for a while - been busy driving 1000km back home.

I have a PixRacer on the way to me and I intend to get it supported in INAV once @martinbudden finishes his work with new IO. Porting new target to Betaflight should be fairly straight-forward. I am eager to try the new ICM gyro :smile:

Maxzor, 2016-07-29:
@digitalentity 
Is the fact that ICM 20608 and MPU 9250 are on the same SPI bus a problem for 8kHz+ interrupts?

digitalentity, 2016-07-29:
@Maxzor no, it won't be the problem - we'll be using just one of them and full SPI bandwidth will be available to that gyro.

Maxzor, 2016-07-29:
@digitalentity I was thinking about EKF and such :) In some places ICM20608 is described as legacy gyro, others as new...

I spent the whole evening looking for a sd detect pin, found none if not SCK.
I was parsing the PX4 source, Betaflight is so much simpler. Glad for now. Were you saying @martinbudden's work on F4 support for BF is not finished yet?

Maxzor, 2016-07-29:
I also came to think a real current sensor could be implemented for this board. It surely is in PX4/APM.

digitalentity, 2016-07-29:
@Maxzor BF anv INAV are getting closer in terms of drivers, one it's finished, targets would be virtually the same between the two. @martinbudden is finishing the new IO and F4 support for INAV at the moment. Hopefully I can already start debugging when my PixRacer arrives.

I'm not sure EKF is fiesible - it's more robust but certainly less understandable and definitely harder to fine-tune.

Maxzor, 2016-07-29:
@digitalentity Can I ask you out of curiosity what will you debug with? Blackmagic stick? I don't have 70€ for a usb+stmf1 stick, but I have FTDI and STML0Nucleo will try to make these work.

Also could use one last precision : is Martin's work with IO and F4 support done on Betaflight? I am sorry to be so BF-leaning I don't know INav yet.

digitalentity, 2016-07-29:
@Maxzor I'm not sure about F4/IO readyness in betaflight, but looks like it's complete and working.

I'm debugging with a cheap chinese STLinkV2 clone like this (http://ru.aliexpress.com/item/Free-Shopping-1PCS-ST-Link-V2-stlink-mini-STM8STM32-STLINK-simulator-download-programming-With-Cover/32329438822.html)

Maxzor, 2016-07-29:
@digitalentity Thank you already! One last question before gn8 : can you build Betaflight on Windows like Visual studio?

digitalentity, 2016-07-29:
Not sure about Visual Studio, I'm using cygwin+gcc for building firmware on Windows.

mikeller, 2016-07-30:
Another option if you want a simple way to build under Windows:

https://github.com/mikeller/docker-betaflight-build

blckmn, 2016-07-30:
Here is how I use Visual Studio:
https://blck.mn/2016/07/setting-up-visualgdb-for-building-betaflight/

Maxzor, 2016-07-30:
@blckmn Awesome thank you, gcc still scares me.
Quite ready to debug :)

borisbstyle, 2016-07-30:
I will try the visual studio for once. Thanks for the manual.

The setting up part is surely easier than eclipse :)
Oh btw do you know how to get all the libraries in it properly?

Missing stdbool.h etc

blckmn, 2016-07-31:
@borisbstyle you mean for intellisense or for the build?

borisbstyle, 2016-07-31:
No for the intellisense of course. It is there in the build.

Maxzor, 2016-07-31:
Do you prefer to work on target by yourselves, quite from scratch, or from a (possibly terrible) pull request from me, that I can do tonight?

digitalentity, 2016-07-31:
@Maxzor If you can do a PR it will be great, I can continue from your sketch and once my board arrives get debugging.

Maxzor, 2016-08-04:
On cygwin compilation I get 3xxKB hex file for Naze target for example. I heard on Raceflight it was dangerous to flash those hex and should rather go for 1MB hex, is it the same in BF?

digitalentity, 2016-08-04:
Size of hex file has nothing to do with actual firmware size. We should be safe.

Maxzor, 2016-08-04:
I have a question (actually many is there a chat for BF? else I can lauch a vanishing one for 5 minutes) about softserial : what is it/is it needed on Pixracer?

digitalentity, 2016-08-04:
@Maxzor, IMO - none. If needed we can add it later.

Maxzor, 2016-08-04:
Sambas greatly helped me providing the timer and pwm arrays.
Is ESC serial to be used?

Maxzor, 2016-08-04:
I am here if you have a tiny moment http://www.disposablechat.com/chat/Betaflight+wonders?password=

Maxzor, 2016-08-04:
I have been adding defines on various files, I am stuck with the timer matters which are as @blckmn  stated the hardest ones.
Some discussion about this would be greatly enjoyed.

Maxzor, 2016-08-04:
I think I have the set of files ready for a test compile.
@blckmn I don't know what you were referring to with "startup file"... stm32f427xx is here?
Anyone up to write the linker script?

Maxzor, 2016-08-04:
Came up with my first PR ( :) ) which is likely to be very terrible.
https://github.com/betaflight/betaflight/pull/893
The pin map should not be too wrong if I can still read...

Maxzor, 2016-08-12:
I have worked on some more files : https://www.dropbox.com/sh/hxgkjs9il27myf9/AACTaWNZSqJy6XpV0JP5UqLHa?dl=0 will not pull now, I still cannot compile.
My VisGDB is likely to be setup wrong, and Cygwin make outputs this http://pastebin.com/73TcDc35.

Maxzor, 2016-08-12:
Scott Shawcroft @tannewt 21:25
@Maxzor #define TARGET_IO_PORTD (BIT(2)) is wrong
its saying you only use PD2
You probably want #define TARGET_IO_PORTD 0xffff

So now Cygwin outputs errors related to target.c only, mainly about the JTAG declaration (target.h) and timer array (target.c). Timers are an unkown realm to me so will take a long time to figure it out alone. http://pastebin.com/fq664aiL

Maxzor, 2016-08-13:
Thanks to the great help of @tannewt, we got a first hex file to debug : you can find it in the dropbox above with the updated files!

digitalentity, 2016-08-15:
https://github.com/iNavFlight/inav/pull/466

digitalentity, 2016-08-15:
The board boots, detects sensors and connects to Configurator via USB. The rest is untested/not implemented yet.

Maxzor, 2016-08-16:
Further test and debug will happen on the Inav PR before coming to Betaflight, BR.

kitsen13, 2016-10-30:
Hello, any news?

borisbstyle, 2016-12-20:
I will close this issue as its not firmware related. The PR for new targets can always be entered.

closed 2016-12-20 by borisbstyle

greybush, 2017-03-20:
is there support for pixracer in betaflight/cleanflight/inav/raceflight?

JiviteshDebata, 2018-04-10:
The link to the schematic is an old one
Please Update to this one since this is the closest one to the latest release
https://github.com/PX4/Hardware/files/93463/PIXRACER-R13.pdf

mikeller, 2018-04-10:
iNav seems to have support: https://github.com/iNavFlight/inav/releases/download/1.9.0/inav_1.9.0_PIXRACER.hex

Support could probably be added to Betaflight as well, if there is interest, and somebody willing to do (potentially a lot of) testing, or provide an engineering sample of the board to a developer.

JiviteshDebata, 2018-04-10:
Yes iNav does have support for it. Currently running iNav on it but everything on the board is not supported yet.
Im currently going to do a lot of testing for it and review it against ardu stack , px4 stack and inav.
As for engineering sample its best not to do that and just buy it and donate it to a DEV 
The hardware revisions have made changes in RC input for Serial RX and PPM so engineering samples maynot have that.

blckmn, 2018-04-10:
I can revisit this.