betaflight/betaflight #852: Cannot set gyro_lpf 188Hz on Dodo, BF 2.9.0

jhznyman, 2016-07-29:
Wanted to try the 'very noisy' setup to see how much it cleaned up the gyro traces. I noticed that setting gyro_lpf = 188HZ caused the pid_process_denom and gyro_sync_denom to default to 1 (8kHz, 8Khz). Didn't notice until I went to fly and the quad wouldn't arm and the configurator was unusually slow when plugging back in. Trying to change the denom values does nothing until gyro_lpf is set to OFF.

This issue seems to have been reported here https://m.reddit.com/r/Multicopter/comments/4qj2f3/betaflight_29_defaulting_to_8khz/ about a month ago as well.

This is not critical for me as the copter flies amazingly well as is. Please close this if it has been reported before or it is user error.

daPhoosa, 2016-07-30:
This is working as intended.  Only for gyro_lpf=off does the sensor have a sample rate of 8kHz.  All other gyro_lpf settings have a sample rate of 1kHz.  So in your situation, both denoms of 1 indicates that gyro and accel are both being sampled at 1kHz.

KiteAnton, 2016-08-01:
This is as intended. Gyro denom and pid process denom is set to 1 whenever gyro_lpf not equal to OFF. 
See https://github.com/betaflight/betaflight/blob/master/src/main/main.c#L623-L626

Could perhaps be a possible "issue/bug" in the configurator that it still shows 8KHz but that should not be addressed here.

closed 2016-08-01 by KiteAnton