betaflight/betaflight #854: Baro mode and my quad goes on full power...

siamect, 2016-07-30:
When I enable baro mode and move the throttle control my quad goes full power. It typically happens when I reduce the throttle. 
I disarmed it in panic and my Quad fell down, luckily hitting a plastic roof so no damage.
 
For testing, I set the max throttle to 1500 to avoid  going in orbit and I held it in my hand. It goes full power and it stops doing that after playing with the throttle control for a while.  
There is also a weird sound from the motors. It has a lower frequency than the motor sound. It's like it the speed was quickly switching between two values. 
I tried to use 1kHz instead of 2kHz that I usually use. The strange sound disappeared but the behavior to go full throttle is still there.

borisbstyle, 2016-08-02:
Maybe airmode should not be used on baro mode perhaps. Baro does different stuff by modifying throttle etc

siamect, 2016-08-02:
This problem is there with or without airmode... I think I had airmode the first time it happened, when I had to disarm the motors to avoid disturbing the aviation business. 
During my later testing, I do not have airmode enabled.

Jordille, 2016-08-02:
Same problem here, it will hunt slightly up and down while slowly loosing altitude in baro mode and if you move the stick above the deadband to climb a little the motors stop, if you move the throttle below the deadband it will launch out of site.

siamect, 2016-08-03:
I thought for a while that maybe the setpoint altitude, which would be close to zero if baro mode is enabled close to the ground, when you reduce throttle it goes to a negative value. If that variable is interpreted as an unsigned anywhere it will go from zero to say -1 but if misinterpreted as unsigned this will be a large value...  
But this doesn't explain the fact that after playing with the throttle, it stops going full speed. Maybe there is an intergrational part of the PID that controls the altitude that is not initialized or that is treated as unsigned?

cnHeider, 2016-08-10:
Same problem

Jordille, 2016-08-11:
It is working for me on 3.0 now, just seems to need some tuning.

RipperDrone, 2016-08-12:
Had same issue of Double Freq Sound on Baro ON before, pls check closed issues. Has been resolved in 2.9+

Regarding full thrust on low throttle or particularly reducing throttle: Happened twice on my SPR3. I think it happened right when I cycled power after upgrading to new Firmware. I got a feel that 

- either quad needs to sit there for a short while to settle the Baro readings/fill the readings queue or
- upon very first Start after Firmware Upgrade, there are some Baro or dynamic PID values needing to stabilizer inflight (is there any Data being stored back depe Ding on Baro values Queue?) 

Spinning up whilst reducing throttle seems to make sense when you hold the quad with your Hand (alt/Vel PID settings try to compensate for the increasing gap actual height vs target height).

Just my thoughts, not sure this helps... Good luck!

siamect, 2016-08-12:
Just loaded 3.0 and sometimes by barometer doesn't work at all. Sometimes it works well. Seems to be an init issue.  
I really haven't tested baro mode yet... (still chicken-shit scared) but one thing I can see, that may be the whole reason for this issue, is that the baro value in my osd goes negative (a few meters) when I give more throttle. So this gives positive feedback probably because of the airflow. So when the baro mode thinks that I'm too low, it will give more power, that will cause the baro to experience pressure drop which will make the fc think we are even lower and give even more power and so on. At some point this will flatten out and the opposite situation will occur, possibly reducing the power to zero. 
Maybe I try to turn the FC upside down and see if I can get this sorted.

RipperDrone, 2016-08-12:
Try foam pad cover on baro

siamect, 2016-08-12:
Thanks I have to review my statement that it sometimes work well... it doesn't, it gives me the derivata of the altitude... I opened a new issue for that.  
I really don't have space for any foam unless I turn my fc around so I guess I will do both...

siamect, 2016-08-13:
Went back to v2.91 and there I still have the Double Freq Sound...  
The baro mode kind of works but is unstable probably because of my baro being exposed to the prop wash. So the motor goes on off and the quad is keeping the altitude but it bounces up and down rapidly maybe with 1-2m amplitude. I adjusted the PID setting for the altitude and it is better but still bounces... I think this is OK keeping in mind that I still don't have any protection around the sensor.
What is not OK is that if I try to move the throttle stick the quad either just drops out of the sky or climbs until I have to kick out the baro mode...

RipperDrone, 2016-08-14:
if baro readings are off, the PID algorithm determines from your spped and (faulty) altitude the corrective throttle value. If inputs are wrong, reaction on throttle will be wrong, too. You might soften the effects by tuning VEL and ALT as per WiKi...

siamect, 2016-08-21:
I'm pretty convinced that the set-point gets completely wrong when I touch the throttle stick. I don't see any indication that it would be any problems with the PID algo being disturbed by a baro sensor that picks up propwash. The propwash I have all the time and there are oscillations because of that. perfectly predictable.  The average altitude is stable until I touch the throttle stick and it either falls out of the sky or it goes full throttle.

siamect, 2016-08-22:
It seems to work on rc8 but not on rc6 which I downloaded from the releases. The rc8 (commit 013bdde1) I built using gcc-arm-none-eabi-5_4-2016q2 on Debian Jessie... I haven't tested that much, but it didn't fall out of the sky but it did go up maybe 5-6m when I enabled baro the first time after power up. After that I could use throttle although it was a bit rough. I have reduced the gain to get rid of pumping but my sensor is still exposed to propwash...

siamect, 2016-10-17:
Cannot test this on the same fc controller board any more because it died from an overdose... 16V into the wrong pin :)  
So i will close this issue and test if this works on my new board when I have time.

closed 2016-10-17 by siamect