betaflight/betaflight #857: Failsafe/Lost Model Alarm not working F3 Frsky X4RSB connected with SBUS; BF 2.9.0

hawaiianer, 2016-07-31:
Hi, I have the problem that the lost model beeping when I switch off my transmitter is not working.
This appears with this F3 board only on BF. With Cleanflight (1.13.0) it is working.
I set the failsafe on the taranis to no pulses before binding.....
Is this a known issue or am I just missing something?

Here is the dump from BF
2016-07-31 @ 10:12:54 -- Running - OS: Windows, Chrome: 52.0.2743.82, Configurator: 1.6.4
2016-07-31 @ 10:12:56 -- Serial port successfully opened with ID: 23
2016-07-31 @ 10:12:56 -- MultiWii API version received - 1.16.0
2016-07-31 @ 10:12:57 -- Flight controller info, identifier: BTFL, version: 2.9.0
2016-07-31 @ 10:12:57 -- Running firmware released on: Jun 25 2016 23:43:54
2016-07-31 @ 10:12:57 -- Board: SRF3, version: 0
2016-07-31 @ 10:12:57 -- Unique device ID received - 0x1e00365233571420343839
2016-07-31 @ 10:13:19 -- Serial port successfully closed
2016-07-31 @ 10:30:39 -- Serial port successfully opened with ID: 27
2016-07-31 @ 10:30:40 -- MultiWii API version received - 1.16.0
2016-07-31 @ 10:30:40 -- Flight controller info, identifier: BTFL, version: 2.9.0
2016-07-31 @ 10:30:40 -- Running firmware released on: Jun 25 2016 23:43:54
2016-07-31 @ 10:30:40 -- Board: SRF3, version: 0
2016-07-31 @ 10:30:40 -- Unique device ID received - 0x1e00365233571420343839
2016-07-31 @ 10:31:14 -- Serial port successfully closed
2016-07-31 @ 10:40:27 -- Serial port successfully opened with ID: 28
2016-07-31 @ 10:40:27 -- MultiWii API version received - 1.16.0
2016-07-31 @ 10:40:27 -- Flight controller info, identifier: BTFL, version: 2.9.0
2016-07-31 @ 10:40:27 -- Running firmware released on: Jun 25 2016 23:43:54
2016-07-31 @ 10:40:28 -- Board: SRF3, version: 0
2016-07-31 @ 10:40:28 -- Unique device ID received - 0x1e00365233571420343839

Setup
Ports
Configuration
Failsafe
PID Tuning
Receiver
Modes
Adjustments
Servos
GPS
Motors
OSD
Race Transponder
LED Strip
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.
$M>$iÜÓÏÛÛÜÛÜÜÜÜÜÜÜÜÜÜÜP$M>?tARM;AIR MODE;DISABLE 3D SWITCH;BEEPER;OSD SW;BLACKBOX;FAILSAFE;$M>ni$M>e¶Ø
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# BetaFlight/SPRACINGF3 2.9.0 Jun 25 2016 / 23:43:54 (3d9b180)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
[... 333 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1500

KiteAnton, 2016-07-31:
Is your beeper working for non-failsafe cases? E.g. Beeper by aux/mode or when arming? 
Does the led on the board blink when failsafe is suppose to be active?

syrekszymon, 2016-07-31:
Same here. SPRacing F3 with Betaflight 2.9.0 and FrSky D4R-II receiver set to no pulse. Failsafe doesn't work.

mikeller, 2016-07-31:
Just an idea: Could be an incompatibility of 2.9.0 with the latest configurator. Can you please check in CLI (`features`) that the feature `FAILSAVE` is enabled?

hawaiianer, 2016-08-01:
Hi, I solved the issue. I reflashed 2.9.0 and DID NOT copy paste my configuration from cleanflight, just configured everything step by step with the Cleanflight Configurator and now it it working!
I compared my working configuration with the one above and I can not find anything different that might have caused the issue. Strange - but the problem was most likely sitting in front of the monitor :-)

closed 2016-08-01 by hawaiianer