betaflight/betaflight #858: Support for Softserial on CC3D REVO.

llambkin, 2016-07-31:
I dont know weather it is possible to implement it, but it would be handy to have, I have a board that I can do some probing of CPU pins if you need.

It would be nice to have it come out on IO port somewhere.

I just want it for FRsky Telemetry. And since the CPU has heaps of headroom, why not?

wildekek, 2016-08-18:
Seconded, the revo support in 3.0.0 is excellent, but soft serial would be awesome.

borisbstyle, 2016-08-19:
Softserial needs a separate timer. @blckmn what do you think of adding this to all fr targets?

blckmn, 2016-08-19:
Sure thing, I can add it to targets as part of the IO work. That way it's easier to specify the pins and timers to be used.

llambkin, 2016-08-20:
Once all the IO pins have been defined and tested, I will make up a picture of the board showing all the pin definitions.

jflyper, 2016-12-15:
Softserial itself works fine without any modification on AIRBOTF4 and OMNIBUSF4s (too bad can't test it with original REVO).

iNav will make use of S5_IN and S6_IN available as tiny pads on these board.

borisbstyle, 2016-12-20:
Closing this one. 3.1 offers the Dynamic IO reallocation.

closed 2016-12-20 by borisbstyle

jflyper, 2016-12-20:
Yeap. An experimental code PR is available as "Configurable serial port pin assignment" #1849.