betaflight/betaflight #863: Latest build hex files

blckmn, 2016-08-01:
It would be good if the latest master build hex files were always available (outside of a zip file).

Or perhaps make them available within the Configurator (i.e. show unstable releases)?

The prior builds can be zipped.

borisbstyle, 2016-08-01:
Unstable releases need to be uploaded to github.

I am planning to do that once the configurator works properly.

AndersHoglund, 2016-08-01:
Will you set up your own CI-engine, or download from my jenkins server? Travis maybe ?

I can set up jenkins to publish .hex, bin and .zip files. Are there a need for .dfu as well?
This however will clutter the artefact page as the number of targets is huge now-a-days. See here what it would look like: http://andwho.sytes.net:8080/job/AndWho_Betaflight/lastSuccessfulBuild/artifact/obj/

Mind you, this is my deskside Ubuntu server. No guarantees on its availability nor up-time.
/A

borisbstyle, 2016-08-02:
I will start releasing unstable releases soon!

borisbstyle, 2016-08-02:
@AndersHoglund 
Not needed to change anything actually. Github has that option already

closed 2016-08-02 by borisbstyle