betaflight/betaflight #864: PWM & PPM at the same time

zdar, 2016-08-01:
I'd like to use PPM & PWM at the same time. The reason is PWM RSSI and/or gimbal control using additional receiver (camera control from another TX&RX).

I went trough the code and if I understand the logic of RX well only small modification should be needed.
- In drivers/pwm_mapping.c after line 300  ppmInConfig(timerHardwarePtr); the channelIndex needs to be increased to map PWMs somewhere in the upper of the channel struccture
- In hardware_maps (in targests) it is neccessary to add into PPM part - unassigned input a pwm entry (like PWM part has)

Is this correct or I'm missing something important somewhere?

Note: This a bit hack. If new system for PWM mapping is in place (issue #712) that could be handled by it.

borisbstyle, 2016-08-02:
isnt this the servo feature?

zdar, 2016-08-02:
No.  Servo feature takes inputs and forwards them to outputs.

This is about input side. i.e. to have PPM and PWM inputs at same time -> more inputs. In theory even two PPM ones (2x8/16 channels).

martinbudden, 2016-08-02:
Could you not just wire your gimbal directly to the PWM outputs of the receiver, bypassing the flight controller altogether?

borisbstyle, 2016-08-02:
Exactly :D. No need for FC there

borisbstyle, 2016-08-02:
@zdar I was having hard time to understand why you would need this actually. But it really doesnt make sense to me. If you have PWM just connect it directly.

closed 2016-08-02 by borisbstyle

zdar, 2016-08-02:
To put it straight PWM RSSI is the most important part about that. Why to make special convertor from PWM->ADC for RSSI when I just few lines and straight connection can do the work...

Gimbal is nice to have (and yes. I have it connected directly nowdays)