betaflight/betaflight #874: See possibilities to gain some efficiency

borisbstyle, 2016-08-02:
@martinbudden 
I know you are good in that stuff. If you have some time please check if there are any efficiency things what can speed up the performance.

For example what I am interrested in is the betaflight pid controller. There are some calculations being done each loop where PIDs and other parameters are being converted to float.
Also the PID scaleres are all extra multipliers.

The calculations are needed for features like inflight adjustments where pids can be changed, but maybe some smart check can be done to notice a change and recalculate only when needed.

martinbudden, 2016-08-03:
@borisbstyle , yes, I certainly think there are efficiency gains to be made.

I'll start looking at this in ernest once 3.0 is released.

gpolic, 2016-09-20:
Hi, I am just browsing here. 
I think the last 2 lines don't have to be inside this loop that you're checking for changed PIDs
They can be moved outside of this.
Also this PID checks could it be done elsewhere or does it have to be so frequent as looptime?

    // ----------PID controller----------
    for (axis = 0; axis < 3; axis++) {
        static uint8_t configP[3], configI[3], configD[3];

        // Prevent unnecessary computing and check for changed PIDs. No need for individual checks. Only pids is fine for now
        // Prepare all parameters for PID controller
        if ((pidProfile->P8[axis] != configP[axis]) || (pidProfile->I8[axis] != configI[axis]) || (pidProfile->D8[axis] != configD[axis])) {
            Kp[axis] = PTERM_SCALE * pidProfile->P8[axis];
            Ki[axis] = ITERM_SCALE * pidProfile->I8[axis];
            Kd[axis] = DTERM_SCALE * pidProfile->D8[axis];
            c[axis] = pidProfile->dtermSetpointWeight / 100.0f;
            yawMaxVelocity = pidProfile->yawRateAccelLimit * 1000 * getdT();
            rollPitchMaxVelocity = pidProfile->rateAccelLimit * 1000 * getdT();

closed 2016-12-19 by borisbstyle