betaflight/betaflight #889: Kiss FC - Configurator Hangs on PID Tuning selection

QuadroBro, 2016-08-04:
Not sure if configurator was updated due to the previous issue with stick input being stuck, but now PID's cannot be adjusted via BF Configurator. I checked in CF to ensure it wasn't a hardware issue and was able to access and adjust PID's in CF without issue. I also reflashed the Kiss FC BF 3.0.0 firmware to see if it had possibly been corrupted, but the problem persists.

In BF Configurator it hangs on "waiting for data". I have given it up to 5 minutes but it never loads and I have to disconnect from configurator to access other tabs.

borisbstyle, 2016-08-04:
Install the new 1.7.0 configurator like release notes say

QuadroBro, 2016-08-04:
@borisbstyle what I was trying to say is that I believe the issue started when it updated to the 1.7 configurator. That's the version I'm running.

mikeller, 2016-08-05:
@QuadroBro: What version of the firmware are you running?

QuadroBro, 2016-08-05:
@mikeller BetaFlight/KISSFC 3.0.0 Jul 18 2016

QuadroBro, 2016-08-05:
Noticed configurator updated to 1.7.1 but have the same problem

mikeller, 2016-08-05:
@QuadroBro : The protocol that is used by the 3.0 firmware was changed between 18/07 and now, and 3.0 builds older than 29/07 are not supported by the GUI any more. You will have to update your firmware, or use the CLI until you update.

QuadroBro, 2016-08-05:
@mikeller Thanks for the info. When I reflashed I used the same build I converted to a dfu. Will grab again and test.

QuadroBro, 2016-08-06:
@mikeller Thanks for the head's up! Problem is resolved.

mikeller, 2016-08-06:
You're welcome, @QuadroBro. Closing the ticket.

deepimpactio, 2016-08-06:
@QuadBro can you share how you converted to DFU?

mikeller, 2016-08-07:
Turns out I can't close it. :-(

QuadroBro, 2016-08-07:
@deepimpactio Download Dfuse Demo which comes with DFU Manager. Open the manager, leave it on the generate bin or hex setting, click the box that says s19 or hex, select the BF hex you downloaded then click generate and name accordingly. For reference, I'm using the version listed in the TBS powercube manual which has worked fine for the Kiss FC

closed 2016-08-07 by KiteAnton