betaflight/betaflight #892: Naze32 rev5 100% CPU not arming

JurikRistjouw, 2016-08-04:
well, above says it all, but it is the gyro/pid in tasks that is about 100% and the copter is not arming. 
Any suggestions? After reverting to 2.9.1 no issues.

[BLACKOUT betaflight_CLI_CURRENT.txt](https://github.com/betaflight/betaflight/files/402885/BLACKOUT.betaflight_CLI_CURRENT.txt)

borisbstyle, 2016-08-04:
you need "set pid_process_denom = 2" for F1 targets on default pid controller

in other words 2k/1k or 1k/1k

or use legacy pid controller with 2k/2k

JurikRistjouw, 2016-08-04:
same on 3rc3 at least.. i am on it

borisbstyle, 2016-08-04:
Just use lower gyro/pid speed ;)

JurikRistjouw, 2016-08-04:
i did, it worked, but also on legacy on 2/2 i had issues which i didn't have on 2.9.1. 

i raised my PID's from default 2.9.1. What is your advice with 1/2khz config with the new betaflight PID? should i revert to you defaults, or try the former settings>

JurikRistjouw, 2016-08-04:
I have now the gyro 2kHz and PID 1kHz betaflight PID and measurement. Changed through GUI for testing purposes (not CLI) All golden again. failsafe everything. Hopfully trying this weekend. Thanx Boris

closed 2016-08-04 by JurikRistjouw