betaflight/betaflight #894: Sparky 1.0 target not working

DaTwo, 2016-08-05:
No serial port detected after flashing to 3.0. Flash back to 2.9.1 works fine again.

nathantsoi, 2016-08-05:
which serial port is broken? or all of them? could you post a screenshot of
the configurator or however you saw it was broken?

DaTwo, 2016-08-05:
I flashed it with dfu 3 times using the bin and the hex on 3.0. I could not get the USB to work as a serial port so you can configure it. I did get some leds to flash like maybe an error. Flashing via dfu to 2.9.1 works flawlessly.

borisbstyle, 2016-08-05:
Are you perhaps using PPM?

Sparky shares uart2 and PPM. Ppm disabled the serial port

DaTwo, 2016-08-05:
This is directly after a dfu flash. Nothing else has changed. It just stops being recognized on the usb port. Flashing back to 2.9.1 fixes it.

MJ666, 2016-08-05:
The Sparky firmware is also not booting on an AlienFlight F3 V1 (this is usually working with loosing some features). This also applies to my F3 development board. Unfortunately I will start traveling next days and can not start debugging the issue. I don't own an Sparky board my self. Looking into the code it looks to be the following lines are missing in the target.h file:

`#define I2C2_SCL                PA9
#define I2C2_SDA                PA10
`

MJ666, 2016-08-05:
Created PR #899.

borisbstyle, 2016-08-19:
Try again

closed 2016-08-19 by borisbstyle

DaTwo, 2016-08-19:
@borisbstyle This needs to be reopened as it is not fixed. The error code via leds is..

blue / red is solid on boot then red / green alternate it looks like 4 fast pause then 2 slow

I am unable to talk see the FC at all on the usb port . 2.9.1 works flawlessly

MJ666, 2016-08-20:
This error sounds like not detecting the Gyro? Kind of strange these the Sparky 1 firmware after the fix with an AF F3 V1 (MPU6500) and my F3 development board (MPU9250).

DaTwo, 2016-08-20:
Would that prevent the usb port from working?

MJ666, 2016-08-20:
IF Gyro is not detected definitely yes.

DaTwo, 2016-08-20:
Is there a location in the docs that tells what these LED codes mean?

DaTwo, 2016-08-21:
@borisbstyle ,Should I create a new issue or can we reopen this? Love to see it fixed before the RC.

MJ666, 2016-08-21:
The fix was only merged to the development branch but not to latest master. @DaTwo what firmware build you actualy tested? It's then also not part of RC8.

DaTwo, 2016-08-22:
@MJ666 , I compiled the dev version and we are confirmed working. My mistake thinking it was in the master.  Thank you!