betaflight/betaflight #897: Motors Would not Disarm

rsjango2, 2016-08-05:
I'm running NAZE 2.9.1 firmware.  I powered my quad with the arm switch in the active position.  No problem.  The motors did not spin.  I had to toggle the arm switch on the transmitter to get the motors to arm.  No problem.  That worked as expected.  After the flight I threw the arm switch to disarm the quad and it remained armed.  I had to flip it over to get it to disarm.  Big problem.

KiteAnton, 2016-08-05:
Please provide more info like dump etc.

borisbstyle, 2016-08-05:
Probably your switch is not configured properly and has problems reaching
disarm state. Thats why you had issues with arming

rsjango2, 2016-08-05:
No trouble with arming.  It was my mistake to apply power with the switch in the armed state.  Once I toggled the arm switch the motors spun as expected.  The problem was at the end of the flight when I went back to disarm.  The motors remained armed.  I had tested this the previous day with the normal procedure where the switch was in the disarm state when I applied power ant everything was fine.  I will include a CLI dump tonight when I get home from work.  Thanks for the help.

rsjango2, 2016-08-05:
# version
# BetaFlight/NAZE 2.9.1 Aug  2 2016 / 22:47:45 (0ef6c3d)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature FAILSAFE
feature RX_PARALLEL_PWM
feature ONESHOT125
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
[... 258 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

#

borisbstyle, 2016-08-06:
Go to modes tab and check how your switch reacts when rx powered. Maybe
your mode borders need to be adjusted

closed 2016-08-19 by borisbstyle