betaflight/betaflight #904: Motors dont ARM on NAZE32 due to 100% CPU (Betaflight 3.0.0-RC4)

ahahn61, 2016-08-06:
Hi,
I just updated to Betaflight 3.0.0-RC4 on a naze32 rev5 board and could not arm the motors.
Actually I observed that in the default setting, with the pid_controller=BETAFLIGHT, the cpu ist loaded 100%.
When I switch to pid_controller=LEGACY everthing is fine again, cpu load drops to 26% (as reported by status and on the status bar of the configurator) and I can arm the motors. 
My experiments where made with the default configuration, except for the receiver which is a PPM (feature RX_PPM).
Any comments?

regards
Alois

with pid_controller = BETAFLIGHT

```
# version
# BetaFlight/NAZE 3.0.0 Aug 5 2016 / 14:31:14 (81c06ac)
# status
System Uptime: 64 seconds, Voltage: 141 * 0.1V (4S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1000, I2C Errors: 0, config size: 1384

# tasks
Task list rate/hz max/us avg/us maxload avgload total/ms
00 - ( SYSTEM) 9 5 0 0.5% 0.5% 1
01 - ( GYRO/PID) 981 886 788 87.4% 77.8% 87321
02 - ( ACCEL) 455 376 167 17.6% 8.0% 19318
03 - ( ATTITUDE) 48 362 326 2.2% 2.0% 2240
04 - ( RX) 48 75 44 0.8% 0.7% 340
05 - ( SERIAL) 24 57182 2 137.7% 0.5% 177
06 - ( BATTERY) 12 40081 1 48.5% 0.5% 68
07 - ( BEEPER) 24 9 0 0.5% 0.5% 6
08 - ( COMPASS) 2 155 124 0.5% 0.5% 79
09 - ( BARO) 24 138 107 0.8% 0.7% 492
11 - ( ALTITUDE) 9 264 152 0.7% 0.6% 273
Total (excluding SERIAL) 159.5% 91.8%
```

with legacy controller (pid_controller = LEGACY)

```
# version
# BetaFlight/NAZE 3.0.0 Aug 5 2016 / 14:31:14 (81c06ac)
# status
System Uptime: 18 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:22%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1004, I2C Errors: 0, config size: 1384

# tasks
Task list rate/hz max/us avg/us maxload avgload total/ms
00 - ( SYSTEM) 9 2 0 0.5% 0.5% 0
01 - ( GYRO/PID) 990 700 609 69.8% 60.7% 11974
02 - ( ACCEL) 988 375 162 37.5% 16.5% 3592
03 - ( ATTITUDE) 96 348 296 3.8% 3.3% 636
04 - ( RX) 49 92 61 0.9% 0.7% 78
05 - ( SERIAL) 96 723 2 7.4% 0.5% 46
06 - ( BATTERY) 48 78 1 0.8% 0.5% 6
07 - ( BEEPER) 98 12 0 0.6% 0.5% 10
08 - ( COMPASS) 9 153 122 0.6% 0.6% 29
09 - ( BARO) 98 134 103 1.8% 1.5% 251
11 - ( ALTITUDE) 39 264 152 1.5% 1.0% 138
14 - ( LEDSTRIP) 98 387 2 4.2% 0.5% 5
Total (excluding SERIAL) 122.0% 86.3%

```

KiteAnton, 2016-08-06:
This is as expected. The new pid controller is more CPU intensive due to additional float calculations. Your Naze32 board is missing a FPU (floating point unit) and therefore all float calculations must be emulated, which cost additional CPU resources.  

Also you seems to be running a lot of sensors, disable those and that will help a bit with the processor load.

borisbstyle, 2016-08-06:
Disable acc or run higher pid_process_denom

ahahn61, 2016-08-06:
SOLVED: with **set pid_process_denom=2** (instead 1) it works fine now

thanx a lot
Alois

```
# version
# BetaFlight/NAZE 3.0.0 Aug 5 2016 / 14:31:14 (81c06ac)
# status
System Uptime: 7 seconds, Voltage: 149 * 0.1V (4S battery - OK), CPU:14%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 1011, I2C Errors: 0, config size: 1384

# tasks
Task list rate/hz max/us avg/us maxload avgload total/ms
00 - ( SYSTEM) 9 2 0 0.5% 0.5% 0
01 - ( PID) 500 708 461 35.9% 23.5% 6580
- ( GYRO) 1000
02 - ( ACCEL) 999 362 143 36.6% 14.7% 2223
03 - ( ATTITUDE) 99 363 331 4.0% 3.7% 475
04 - ( RX) 49 78 44 0.8% 0.7% 43
05 - ( SERIAL) 99 702 2 7.4% 0.5% 9
06 - ( BATTERY) 49 40085 1 196.9% 0.5% 45
07 - ( BEEPER) 98 10 0 0.5% 0.5% 2
08 - ( COMPASS) 9 130 98 0.6% 0.5% 17
09 - ( BARO) 98 138 105 1.8% 1.5% 177
11 - ( ALTITUDE) 40 270 157 1.5% 1.1% 104
14 - ( LEDSTRIP) 98 428 259 4.6% 3.0% 388
Total (excluding SERIAL) 283.7% 50.2%

#

```

closed 2016-08-06 by KiteAnton