betaflight/betaflight #915: PPM rx does not work on REVO

totaltmega, 2016-08-07:
Hi, have the same issue here as with racefllight. https://github.com/rs2k/raceflight/issues/195
tried both pin 10 and pin 5 of the Flexi-IO connector.
Last known working condition of PPM is raceflight RC10

Even tested master today, related to the bug #812 ?
Let me know if you need me to test something.

blckmn, 2016-08-07:
Hi there. If you're willing to test I'll help work through the issue with you.

totaltmega, 2016-08-07:
Yes please :=) We can start with the pin used for PPM on Revo, is it Flexi-IO Pin10 (as it is in Raceflight RC10) or Flexi-IO pin 5 as in LibrePilot: http://opwiki.readthedocs.io/en/latest/user_manual/revo/revo.html

Thanks

martinbudden, 2016-08-07:
If the fix involves changing pin assignments, can we change them to be compatible wit LibrePilot rather than Raceflight?

This would allow users to change to and from LibrePilot without rewiring.

blckmn, 2016-08-13:
@martinbudden have now got a PR for this https://github.com/betaflight/betaflight/pull/993 the pin is the pin #5 on the FlexIO as per librepilot.

This is tested with Spektrum receiver and D4R and works a treat for me :)

blckmn, 2016-08-13:
@totaltmega should be working now for you too if you can let me know once my PR is merged.

totaltmega, 2016-08-13:
Hi, im sorry but it doesn't work for me. Pulled the latest master and built the code checking that your changes were in. Resetted the config, changed to PPM as rx, disabled the additional UARTS, rebooted.

I changed to pin5 and the receiver tab just shows the default values as on the pic:
[image]

I even tried to build the code with the PWM1 changed back to PWM6 to see if Flexi-IO pin 10 worked but with the same result. I dont know why yours working but not mine, as Revo Mini schematic should be identical to original Revo. I even checked with oscilloscope to see that my receiver still outputs the right signal.

Anyone else with at Revo mini that can try PPM?
Anything else I can try or debug somehow?

Best
Christoffer

llambkin, 2016-08-13:
I have a friend with the same issue. This makes things hard because I cant directly test it.....

roninnn, 2016-08-14:
same here, no connection to rx on pin 5 or 10.

blckmn, 2016-08-14:
Can you please photograph your setups indicating which pin where etc. it's working fine here, both with taranis d4r and orange (spectrum)

roninnn, 2016-08-14:
RX: FrSky D8R-XP Pin 3+4 jumpered for PPM
On the photo RX: Black, Red, White -> PPM. Where White goes to flexiIO Receiver Port #5

http://de.tinypic.com/r/15zfwk3/9

Let me know if you need more information

aussiepage, 2016-08-18:
I compiled and tested latest master today but PPM did not work for me on either the input used by OpenPilot or the one used for Raceflight. The compile was for REVO target with OPBL enabled. Downloaded via openpilot tools using rescue option. I tested with D8R ad D4R receivers. The revo is an original and is stand alone (not in quad) with only receiver connected. Here is a link to the bin file produced by my environment - https://dl.dropboxusercontent.com/u/2479293/betaflight_3.0.0_REVO.bin. 

Would be keen to hear if blckmn or others could test this in working setup to verify if the difference is board or build environment related? Would also be good to test a bin file from others who have PPM working with my hardware.

Thanks to all working to solve this issue - I will see if I can help.
Bruce

blckmn, 2016-08-18:
I have a working RTFQ Revo (full size) here, using both a D4R and a Spektrum (OrangeRX) to be sure - I can upload the video if you like. I will double check the master branch tonight and make sure there has not been any issues reintroduced.

blckmn, 2016-08-18:
To confirm please do not bother testing the pin 10 - only pin 5 is configured to be used.

aussiepage, 2016-08-18:
Many thanks - would appreciate it if you would do a quick test with bin
file I shared on Dropbox. Also are you uploading via open pilot boot
loader? Could see your changes in master but not working for me :( Must be
close give you have working configuration.

Thank heaps
Bruce

totaltmega, 2016-08-18:
Hi, I built master today as well without improvement.
Here is screens from the PPM signal going to pin 5. Its 8ch and the spec is correct according to the last pic, so it should work.

[image]
[image]

[image]

Best
Christoffer

borisbstyle, 2016-08-19:
@blckmn 
Perhaps this timer needs to be removed 
https://github.com/betaflight/betaflight/blob/master/src/main/target/REVO/target.c#L32

blckmn, 2016-08-19:
https://github.com/betaflight/betaflight/pull/1043

[image]

aussiepage, 2016-08-19:
@blckmn 
#1043 changes worked for me :) - thanks

roninnn, 2016-08-19:
changes work for me too. great :-)
signals are correct, but now motors won't spin after arming.
Anyone else having this, or am I missing something?

totaltmega, 2016-08-19:
>  aussiepage commented 3 hours ago

@blckmn
#1043 changes worked for me :) - thanks

Me too!
Thanks for the fix!

closed 2016-08-27 by blckmn

azrealvx, 2017-05-18:
Hi guys.
I am a complete noob and facing this issue with the receiver. How do i go about implementing this fix?
Thanks for the help!!

mikeller, 2017-05-19:
Hi @azrealvx. This issue was fixed a long time ago, so unless you are using really old firmware (and should update it), it is unlikely that your problem is caused by this.

This seems to be a configuration problem, please consult the RCGroups thread (https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread) to check for configuration tips, or ask there.