betaflight/betaflight #916: Support for configuring of 5.8GHz modules via SPI

MartinLaza, 2016-08-07:
Modules based on RTC6705 like TX5823 could ve configured via SPI. This would lead to simple VTX without any additional electronics on the VTX. Selected channel shall be selected via channel from transmitter and forwarded to telemetry data. Than this could be displayed on ie. Taranis display.

IB1387, 2016-08-10:
The Unify Pro Lineup has also the ability to control the VTX over a protocol called SmartAudio. Information can be found here: http://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf
To get technical documentation, one can create a ticket here: http://team-blacksheep.freshdesk.com/

Would be very nice to have a solution for many of the available VTXs.

SoulPictures, 2016-08-10:
it seems like someone has already done this for "Dronin" maybe it can be ported from there: https://github.com/d-ronin/dRonin/tree/next/flight/Modules/VTXConfig

nathantsoi, 2016-08-13:
this was already implemented by @savaga, check the sirinfpv target :)

rdarby777, 2016-08-17:
Im not sure I follow...I am using a MotoLab board.  How can I use this feature on that board.  Has it been tested with the Unify Pro HV or any of the FC's

NightHawk32, 2016-08-28:
This is also discussed in #1029 .
But don't mix RTC6705 and SmartAudio. SmartAudio uses single wire UART whereas RTC6705 is controlled via SPI. RTC6705 support is already implemented but vtx abstracion is not complete. It's more or less only usable on SINGULARITY or SIRINFPV targets.

jflyper, 2016-08-28:
RTC6705 bit-banging can be done with any target with 3 free gpios. 
See cleanflight/#1948 for basic implementation.
Also see [https://github.com/jflyper/cleanflight/tree/vtx-integration-with-vtxcustom](url) for a version with stick command compatible with SINGULARITY.

And watch out. @sblakemore was talking about RTC6075 is ill-behaving as a SPI device; it listens to MOSI regardless of the SS state. (My guess is that SS is only used for 'framing'.)

closed 2016-12-19 by borisbstyle