betaflight/betaflight #919: What are the "representative" targets for development/testing?

martinbudden, 2016-08-07:
Before making a PR it is good practice to do a `make all`. However with all the new targets, this now takes ages. It would be useful to a have a representative list of targets, that can built before making a PR.

Any views on what these should be? The list should contain F1, F3 and F4 targets.

borisbstyle, 2016-08-07:
It would indeed be nice to have a build control system on PR before its
merged. Something like cleanflight has with travis.

Before it was much easier to do build all.

Now I always check smallest targets like CC3D and also ine f3 and f4 target.

AndersHoglund, 2016-08-07:
Cleanflight uses Travis that builds, and I beleive also runs the unittests, on each PR before integration to master/dev/rel/maint/int -branches. This way a failing PR-build is not a blocker and it is safe to just commit a PR and let Travis ( or some other CI-engine) do the legwork. Not sure if there also is a notification system to the PR-issuer, that would be handy.
Travis seems to work hand-in-hand with github nicely, my Jenkins does not really..... Only checks after integration, and only polling for updates every 15m.  
I don't know anything about Travis and why the Cleanflight methodology was dropped.

martinbudden, 2016-08-07:
Yes, in Cleanflight Travis builds and runs the tests.

I think it is a good system overall. The only slight problem with it is that it runs on PRs sequentially, so if there are a few PRs "in the queue" before yours, it can take a while before you get to see the results of your PR.

This should not normally be a problem, but with Cleanflight there tends to be a flurry of activity before a release, so there are lots of PRs. My view is that is bad development practice anyway - the flurry of activity should be just after a release, not just before one.

AndersHoglund, 2016-08-07:
Agree. Or actually "CF" (Continuous Flurry). Make a release branch (for small release fixes) and carry on as usual with flurry on master, not disturbing the releases nor being disturbed.
My understanding was that Travis runs each PR in parallel with other PRs. No serialization. But I might be wrong. And of course if things are added to an existing PR, there will be serialization.
What I have not found in Travis is the wast amount of add-ons available in Jenkins for a variety of things, but then again: I don't know Travis that well to rule it out..

AndersHoglund, 2016-08-08:
Made a couple of trails on Travis, using betaflight/master. Only a few hacks needed in the .travis.* files we already have. See my repo here: https://github.com/AndersHoglund/betaflight/commits/travis_trail
The results here: https://travis-ci.org/AndersHoglund/betaflight
Needs some more polishing and a lot more understanding how it works and where the artefacts are, how/where to publish etc etc... 
All current 45 targets took some 35m to build, travis seems to run 4 - 5 targets in parallel.

borisbstyle, 2016-08-08:
Nice man! You are fast :)

borisbstyle, 2016-08-08:
travis.yml file needs to be changed I see

martinbudden, 2016-08-08:
For Cleanflight travis builds targets in parallel, but PRs in series. I don't know if it can be set up to do PRs in parallel as well.

35 minutes is a long time to wait, so perhaps Travis should only build a representative subset of all the targets?

AndersHoglund, 2016-08-08:
Oh, yes. I have only changed what is absolutely needed for a trail run. 
But I don't like that hard coded target list...., not sure how to auto-generate it though. From "make targets" I would have hoped. But this list needs to be there before everything starts.. hen&egg.  
But if you just want a few target samples, it is great.

35m is surprisingly slow.... But it is a free service.

borisbstyle, 2016-08-08:
@martinbudden 
There are not that much PR's in betaflight. It is still good to have it on PR's for the reviewers so they quickly can see if it all builds. Usually the review happens long time after PR is applied.

AndersHoglund, 2016-08-08:
@borisbstyle If you get an account on Travis connected to github "betafight/betafligt" then I can PR the travis.* files as they are. Then we can continue refining things from this starting point.

borisbstyle, 2016-08-08:
Ok let me check how to get that

borisbstyle, 2016-08-08:
@AndersHoglund betaflight account is accessible

AndersHoglund, 2016-08-08:
Easiest via your github account "View profile and more" (top right corner) then "Integrations". There you find Travis. Two clicks (or so..) later you have a Travis account.

AndersHoglund, 2016-08-08:
OK, lets commit a PR. I can do one with the .travis files on the "development" branch. Stay tuned.

AndersHoglund, 2016-08-08:
You need to flip the switch for repo betaflight/betaflight. I can't do that it seems.... Travis says:
    This is not an active repository

AndersHoglund, 2016-08-08:
on this pabe: https://travis-ci.org/profile/betaflight

borisbstyle, 2016-08-08:
How can I do that?  Not seeing many options there at all

borisbstyle, 2016-08-08:
Ok try again

AndersHoglund, 2016-08-08:
Yeah!! It's running: https://travis-ci.org/betaflight/betaflight

AndersHoglund, 2016-08-08:
Made one more PR. Both seems to be running in parallel.
https://travis-ci.org/betaflight/betaflight/pull_requests

AndersHoglund, 2016-08-08:
Nope. Looks as if they are both running on that  page. But one is stalled, waiting.

martinbudden, 2016-08-08:
@AndersHoglund - great stuff!

borisbstyle, 2016-08-08:
Awesome!

AndersHoglund, 2016-08-08:
Thnx guys, but I am only re-using and starting what was already there from the beginning.

borisbstyle, 2016-08-08:
@AndersHoglund 
How do you download the builds actually?

AndersHoglund, 2016-08-08:
Not sure there are any artifacts saved at all. I think there is a need for doing uploads to github/releases via scripting to do that. Will look into that, in meantime master and development artifacts are here as usual: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/ and http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/

borisbstyle, 2016-08-08:
@AndersHoglund You could use bintray I see. https://docs.travis-ci.com/user/deployment/bintray. Any experience with that?

AndersHoglund, 2016-08-08:
Nope, will look into bintray.

AndersHoglund, 2016-08-08:
Noticed we are overloading Travis already. lol... 5 builds on queue ..
Funny thing is that we can get up to 3 builds on the same PR. First on the PR itself, then one build when it 's merged to development/master. And finally a 3:rd build when the development commit goes into the #879 PR. IWill check if it is possible to block certain PRs in Travis.

AndersHoglund, 2016-08-09:
Travis is not up to the task. 
Currently 4 PR builds and 7 master/dev/RC builds in que. PR builds seem to have prio and blocks (?) the other. Funny things that Travis created a master build and a RC build dispite the two are on exactly the same revision (366fb7e). One build is clearly enough. Travis could have attached the RC-label only...

martinbudden, 2016-08-09:
Yes, it is totally overloaded. And as you mention, it doesn't seem to have the intelligence to avoid repeating work.

However it is still beneficial. We could run it on a subset of the whole build - that would still catch some problems, and would be better than not running it at all.

So we seem to have come full circle - back to my original question: what is a representative subset? How about:
```
CC3D NAZE
ALIENFLIGHTF3 SPARKY SPRACINGF3 SPRACINGF3EVO STM32F3DISCOVERY
ALIENFLIGHTF4 BLUEJAYF4 FURYF4 REVO
```

AndersHoglund, 2016-08-09:
Well, all (or almost all ) targets are unique in some way. If we leave one out, Murphy says that one has a problem and would have failed. 
But lets try your list and see if Travis can cope with that.  I would like some small OPBL target as well, usual culprit.

borisbstyle, 2016-08-09:
Just add of eacht type 1 and add the special ones like cc3d for example.

The ones with least space etc

martinbudden, 2016-08-09:
I've just cancelled a bunch of builds. The only ones left in the queue are the "shorter" ones, so hopefully travis will catch up now.

AndersHoglund, 2016-08-09:
Now we are down to about 8 to 10min for 12 targets. Downside is that we will not catch all target specific build problems. 

I will not continue any research on some free cloud  based ARM, like bintray etc. KInda meaningless to only publish 12 out of 45 binaries. 
Might pick it up later on and maybe some optimization and caching etc will make Travis perform so all targets can be built and published. Until then Jenkins rulez.  Someone else that knows Travis and/or bintray (or some other CI / ARM systems) better are welcome to have a go at it.

AndersHoglund, 2016-08-09:
One good example where a PR is too big for a target, one failing out of 12. Guess what target...
https://travis-ci.org/betaflight/betaflight/builds/151044269
There is another linkage problem on another target as well.. 

Both those shows a strength of Travis over Jenkins as it is setup now, direct overview where tha problems are and easier to browse the buildlogs per target than one long 45 target log .

mikeller, 2016-08-09:
I think having travis do the 'representative' targets and integrate the results with the PRs is still a lot better than how it was before, at least this should catch the majority of error cases, and display the results right in the PR, thus making it easy to do the right thing.

AndersHoglund, 2016-08-09:
Ooh yes. The presentation of Travis results in GitHub is really good. And especially Travis doing pre-checks on PRs before merge is invaluable. 

If only Travis could run a little faster. Must be possible to have a cached pre-configured VM to run in, now each job installs stuff in a new VM for each job/target. Rebuilding the same VM over and over. In the Travis docs there is a recommendation NOT to cache for example debian package installations.. hm strange. https://docs.travis-ci.com/user/caching/#Things-not-to-cache

AndersHoglund, 2016-08-09:
Aha, Travis caches are off-site tar-balls on S3. Why the h.....? Disk space is cheap. 
https://docs.travis-ci.com/user/caching/#How-does-the-caching-work%3F

closed 2016-08-19 by borisbstyle