betaflight/betaflight #939: Sparky 2.0 Accelerometer

Tomcatt, 2016-08-09:
Got my hands on a Sparky 2.0 and wanted to give it a go with BetaFlight.  The Accelerometer indicator  is not working so I can't calibrate the board.

borisbstyle, 2016-08-09:
You dont have to calibrate for acro modes ;)

But someone will check that probably

borisbstyle, 2016-08-09:
@MJ666 Can you look into this since you did the https://github.com/betaflight/betaflight/pull/900

MJ666, 2016-08-09:
I can take a look at this when back home. @Tomcatt we should get in direct contact to proceed. I don't own an Sparky 2 board. It does not make sense to me if the Gyro is properly working and the Acc is not? Please can you check the sensor activity of the sensor tab.@x-sergo is reporting the Sparky 2 is even not flashing properly #891?  May be also @frandiazc can also help testing.

frandiazc, 2016-08-09:
I already installed raceflight....and raceflight install RFBL, other bootloader and then, when i install this version, doesnt work....only green led on....shit

MJ666, 2016-08-09:
If you use DFU mode (short SBL pads during power up) in the BF configurator it schould not make any difference what was installed before. Now I have 2 reports of an not working software and one with an failed ACC detection. @Tomcatt how did you flash the Sparky 2?

skullq, 2016-08-09:
@MJ666 
How can i help you?
I have spark2 with latest bf.
# version
# BetaFlight/SPARKY2 3.0.0 Aug  8 2016 / 20:12:27 (72154c3)

frandiazc, 2016-08-09:
Just installed betaflight, and I have the same problem. Acc appears unabled.

skullq, 2016-08-09:
I think, MPU9250 is not in the list.
acc_hardware = AUTOAllowed values: AUTO, NONE, ADXL345, MPU6050, MMA8452, BMA280, LSM303DLHC, MPU6000, MPU6500, FAKE

Tomcatt, 2016-08-09:
@MJ666 & @x-sergo Quick background:  I thought being an upgraded REVO, I flashed it like one and it worked until I reboot.  Started do some digging and as ways in the we,we hours of the night I found a link that got it to work.

You from what I can tell, you CAN flash it in DFU mode with .hex file but it will NOT boot up.  I think the data is being written to another location on the board.  The link says to have it converted to .dfu file and THEN you can flash it.  No extra port usage all single USB.

http://dronehitech.com/flash-raceflight-sparky-2-0/

You'll have to download 2 files one will convert the .hex to .dfu file.  The .bin you MUST use the 08000000 address on the main flash RAM area.  No you can't use the old way with the Flash Loader Demo app because it wants a COM port not a bootloader in DFU mode.

DfuSe Demo will see the DFU for you.  This app will upload to the board from a .dfu file use converted from the app DFU File Manager.

All these files you'll get from the link above.

Hope this helps.

I'll test out the other options on the ACC issue.

Tomcatt, 2016-08-09:
@MJ666 No Joy on the manual setting of the ACC set command and FAKE setting didn't work too.

Tomcatt, 2016-08-09:
UPDATE: I got the .hex to work.  Had to go back and I see you guy added SPARKY2 to the MAKE compile list.  I'll email the rest.

Tomcatt, 2016-08-09:
I’ve also installed RF.  It took some time to figure that all out too.  Please look at the post; I uploaded a link but I have some notes there that the key to make it work from the .bin file.

AndersHoglund, 2016-08-09:
MPU9250 is listed in the CLI serial_cli.c as if it is a gyro only MPU. 
But it has Gyro, ACC and MAG and should be listed in three string sets, not only in one. 

    static const char * const sensorHardwareNames[4][11] = {
        { "", "None", "MPU6050", "L3G4200D", "MPU3050", "L3GD20", "MPU6000", "MPU6500", "MPU9250", "FAKE", NULL },
        { "", "None", "ADXL345", "MPU6050", "MMA845x", "BMA280", "LSM303DLHC", "MPU6000", "MPU6500", "FAKE", NULL },
        { "", "None", "BMP085", "MS5611", "BMP280", NULL },
        { "", "None", "HMC5883", "AK8975", "AK8963", NULL }
    };
The ACC and MAG detection code would also need to be checked

skullq, 2016-08-09:
MPU9250 is not only gyro MPU. It contains ACC and Compass too.
"MPU-9250 Nine-Axis (Gyro + Accelerometer + Compass) MEMS MotionTracking™ Device"

From BF serial_cli.c
static const char * const sensorHardwareNames[4][12] = {
    { "", "None", "MPU6050", "L3G4200D", "MPU3050", "L3GD20", "MPU6000", "MPU6500", "MPU9250", "FAKE", NULL },
    { "", "None", "ADXL345", "MPU6050", "MMA845x", "BMA280", "LSM303DLHC", "MPU6000", "MPU6500", "MPU9250", "FAKE", NULL },
    { "", "None", "BMP085", "MS5611", "BMP280", NULL },
    { "", "None", "HMC5883", "AK8975", "AK8963", "MPU9250", NULL }
};

Yes, some code are required to check MPU9250.

AndersHoglund, 2016-08-09:
Acc detection is not used in sensor/initialisation.c . Will do a PR fixing this.

MJ666, 2016-08-09:
The MPU9250 driver is actually only used for the Sparky 2.0 and REVONANO targets. very likely there is still on isue there. On option wild be to change to the MPU6500 SPI driver since the GYRO/ACC part is the same. The MAG driver is separte anyhow and is actualy only working well for F3 target. There is still an issue for the F4. Likely some timing problem.

AndersHoglund, 2016-08-09:
PR #944

Tomcatt, 2016-08-09:
YES!  ACC now online!!

Thanks

frandiazc, 2016-08-09:
Now acc is working but its appears to be slow.....slow movements and look is not working well

borisbstyle, 2016-08-09:
@frandiazc 
That has usually to do with something else. MAG or some other sensors

frandiazc, 2016-08-09:
@borisbstyle And the solution is?? i turn it to right, and its moves crazy!

frandiazc, 2016-08-09:
Only the fc connected to usb

frandiazc, 2016-08-09:
Try to disable mag and baro and the same, it moves erratic.

Tomcatt, 2016-08-09:
REVO wasn’t slow but I’ll turn them off and see what happens.

AndersHoglund, 2016-08-09:
Fixed the CLI "Allowed Values" printout in PR #946

Tomcatt, 2016-08-09:
I don’t know, loop time stayed the same.  Mag and baro are off.  I guess it’s a toss up.

Tomcatt, 2016-08-09:
Compiling now…

AndersHoglund, 2016-08-09:
MPU9250 mag support still missing. Who needs that ?

Tomcatt, 2016-08-09:
Dump still crashes.
> On Aug 9, 2016, at 23:52, Damon Keller <bluecheese@reagan.com> wrote:
> 
> Compiling now…

Tomcatt, 2016-08-09:
I was using it in flights along with the baro.
:)

AndersHoglund, 2016-08-09:
CLI dump are other issues, #945 and #943. 
MPU9250 MAG support should perhaps be new issue as well, so we can close this one. If there is an urgent need.

borisbstyle, 2016-08-09:
You can use "diff" instead of dump

Tomcatt, 2016-08-10:
Bro, it still locks up when I issue the command.

MJ666, 2016-08-10:
The AK8936 driver is for the integrated MAG of the MPU9250. It is actualy not working for F4 targets. There is likely an timing isue I was not able to fix yet. May be the mag should be disabled by default for this targets where this apllies.? Thoughts.

Tomcatt, 2016-08-10:
Are you able to build 2 drivers for it and give us the option to turn the MAG on or off?

MJ666, 2016-08-10:
Don't understand what mean? Why need 2 drivers?

Tomcatt, 2016-08-10:
It sounds like you’d need to have a driver for the mag and acc.  Unless you can work both devices under one driver.

I’d like to give you guys a hand in coding but that’s not my area.

MJ666, 2016-08-10:
The MPU9250 is an combination of an MPU6500 (Gyro/Acc) with an AK8936 (Mag) chip in the same case. The Gyro/Acc is covered by a single driver (mpu6500 or mpu9250). This is actualy kind of redundant in BF. The Mag part is covered by an seperate driver (ak8936). The connection inside the chip is via an I2C bus. This is not very efficient and impacts performance. The comunication also is tricky and looks to be very time critical. In my view its not a good plan to design SPI based FC's with the MPU 9250 for this reason. An seperate Mag sensor is recomended.

Tomcatt, 2016-08-10:
Your in down in the weeds of it.  :)

Well, then I guess well go without the mag then.  Sounds like a weekend project to get it all running right.  I didn’t know the Sparky 2.0 had combined some of the parts into one.  If I remember the REVO had separate parts right?

x-sergo, 2016-08-10:
@Tomcatt @MJ666 Have you managed it to flash the Sparky2.0 with the Betaflight and get it connected to the Configurator? Mine is only showing the green LED on, as it is constantly in bootloader mode.

@Tomcatt Thanks for sharing the link. It is actually my own blog lol

Tomcatt, 2016-08-10:
@x-sergo
LOL
I didn’t know. 

No, I wasn’t able to flash it with the Configurator hens why I use longer way around to flash it.

The only thing I can think of is that it’s now point to the right address when flashing.  Yes it WILL flash but where’s going I don’t know.

I tried using MacOS CLI way to flash it with the dfu-util command and it give me an error.  I can’t remember it and I’ll have to play with it after I come home from work.

MJ666, 2016-08-11:
I don't own an Sparky board. Flashing should work with the configurator. You may need to reduce the baudrate from default. Force the board to DFU mode via shorting the SBL pads. You need to replace the STM DFU driver with zadig. There is an flashing guide for VCP targets in the CF documentation folder. The process is really the same for all VCP based boards.

AndersHoglund, 2016-08-11:
I have to use RFC 1.3.5 flasher on Revo. 
BFC flasher only hangs in "Erasing" in DFU mode. Issue on GitHub BFC created.

Tomcatt, 2016-08-12:
Try slowing down the baud rate.

tianbin4279, 2016-08-22:
@MJ666  Can you tell me your email do?

BayaroX, 2016-09-11:
Is it right, that das is "only" a optical problem? The quad wont show moves over yaw-axis and the other axis shown very slowly/freezing in configurator - but while flying this is not a problem? So its possible to fly with sparky 2.0? Cause we have the same bug with our 3 sparky 2.0.
We use USB VCP to connect sparky 2.0 to computer.

closed 2016-12-19 by borisbstyle