betaflight/betaflight #945: Sparky 2.0 CLI Dump

Tomcatt, 2016-08-09:
Found another bug.

Trying to make a copy of my CLI dump and it locks up as soon as I hit the enter key.

AndersHoglund, 2016-08-09:
There was a string array size problem in Jenkins build 496 (9ae06cd) Try 497

AndersHoglund, 2016-08-09:
Ouch. Not only Sparky2. My Revo on (da9673b) also hangs on the dump cmd.
There was multiple issues committed here, cli diff and acc fixes. Not sure which to blame. Will bisect...

AndersHoglund, 2016-08-09:
Dump cmd hangs already on commit (78c2da5) , PR #943

borisbstyle, 2016-08-09:
I am now testing on F3 targets and it seems fine there :D

AndersHoglund, 2016-08-09:
Commit (271eb4c), prior to PR #943, works fine: 

    # dump
    
    # version
    # BetaFlight/REVO 3.0.0 Aug  9 2016 / 17:13:48 (271eb4c)
    
    # name
    name -
    ......... etc etc

AndersHoglund, 2016-08-11:
Dump and diff still hangs on Revo. Despite added delays and what not.

     # version
     # BetaFlight/REVO 3.0.0 Aug 11 2016 / 16:00:57 (bf679b6)
    
    #

AndersHoglund, 2016-08-11:
Yes! Commit (0d057dd) seems to fix this for Revo.

borisbstyle, 2016-08-11:
Fixed indeed

closed 2016-08-11 by borisbstyle

Tomcatt, 2016-08-11:
Sparky 2 is a go, thanks.