betaflight/betaflight #948: Sparky 2: Serial ports not working

hrrr, 2016-08-09:
Hi!
Trying 3.0.0-RC5.0. All serial ports appear as UART1, and it is not possible to change them:
[image]

Anybody else with the same issue? Or is my board buggy?
Thanks!

Tomcatt, 2016-08-10:
Yes, you can not set the ports up.  It will revert back to default settings.  Let me install RF so you can see what it looks like.

Tomcatt, 2016-08-10:
[image]

borisbstyle, 2016-08-10:
The problem is easy to solve. @MJ666 used the old definition of serial ports instead of the new one :)

@mj666 should be UART and not USART
https://github.com/betaflight/betaflight/blob/master/src/main/target/SPARKY2/target.h#L85

MJ666, 2016-08-10:
My bad. Will make an related PR to fIx this.

Tomcatt, 2016-08-10:
S.Bus port works good.  My FC I had to remove what looked like a resistor on a jumper, brigde it so I can have 5V go to the receiver.

Also, I need to test the OSD but I haven't built one for this quad yet.  I may have to still it from my Mark 2 quad.  LOL

MJ666, 2016-08-10:
Some clone Sparky boards come with resistors solder to the two RX power solder pads. Best to remove both an make the proper solder bridge.

Tomcatt, 2016-08-10:
That was easy, but doesn't help to know I own a clone.  Now the CAP8 on the baro needs to go to stop it jumping so much.  The REVO needs it too but the board I have is too new and can't find any docs on it.

frandiazc, 2016-08-10:
@Tomcatt can you check your model movement in the guy? I have problem, it moves erratic...please..

hrrr, 2016-08-10:
Serials working fine with the new code! Thanks guys!

closed 2016-08-10 by hrrr