betaflight/betaflight #955: KISS FC - Add SmartPort Telemetry on AUX1 pad

basdelfos, 2016-08-10:
By default on the KISS FC firmware assigns SmartPort telemetry to the AUX1 pad.
Running Betaflight 3.0 on a KISS FC requires to solder the SmartPort wire over to the TX pin from UART1 or UART3. For ease of switching between Betaflight and KISS fw it would be easier to assign SmartPort telemetry to AUX1. Maybe, softserial could be used for that (if available on the KISS FC).

basdelfos, 2016-08-11:
[image]
Copied from another issue, image is send to us from the KISS headquarters

basdelfos, 2016-08-11:
I made the changes last night but did not yet succeed to get SoftSerial1 on AUX1/PA13 and AUX2/PA14 (resp. TX/RX). I think the problem is with the different timers these pins use. Pin PA13 uses TIM4 and PA14 uses TIM8 according the SMT32F303 datasheet. I had the suspicion this could be a problem and turns out to be true (or something else is not right yet). 
If so, I will have to find another pin for SoftSerial1RX that uses a channel on the same timer as PA13 and is available. Pins PA11 or PA12 are candidates with timer TIM4, but they are labeled D- and D+ and I don't know where that routes to. When I use one of these pins for SoftSerial1RX, could I make a short and fry my KISS FC?

Would it be possible to only use the TX on a SoftSerial and disregard the RX? That could also solve the problem. But then this SoftSerial port is only usable for telemetry information.

I assigned SoftSerial2 to pins CAN_RX/PB8 and CAN_TX/PB9, they use both a channel on the same timer TIM8. Did not test it yet tho.

Maybe @llambkin can help with answering these questions?

llambkin, 2016-08-11:
Sorry @basdelfos, I am no programmer, I just tinker about with others work, I think, much like you are doing. As far as I know you need both RX and TX shorted for softserial to work. So I have no idea how KISS implemented it and got it working, because as far as I knew you needed a UART that is inverted to make s.port to work.

Maybe someone with more F3 experience could clarify? How did KI$$ FW make it work off one pin? Can AUX1 be an actual UART? Or have they made some sort of voodoo hack to make it work off one pad only?

borisbstyle, 2016-08-11:
@llambkin Softserial supports inversion too. You can get smartport even on F1 boards with RX/RX on same pin

borisbstyle, 2016-08-11:
@basdelfos 
If this is a SOFTSERIAL it can be added to this pin, but requires testing. I honestly think it is more correct to just use UART.

llambkin, 2016-08-11:
Yeah and it frees up CPU time for extra betterer things like superexpo and fancy new filters and such.

llambkin, 2016-08-11:
@borisbstyle, is it just classic old ppm rx's that require both pins shorted?

borisbstyle, 2016-08-11:
No if you want to use Smartport on softserial you need to short TX and RX pin from softserial to get smartport working. Something like this here
https://oscarliang.com/sbus-smartport-telemetry-naze32/

I have a KISS FC now too, which I put on one of my quads and loaded with 3.0. It really is no big deal to put the telemetry wire to the TX of UART3, which is on the side of the board

basdelfos, 2016-08-11:
@borisbstyle 
According to the F3 datasheet, there is no UART_TX available in the function table for pin PA13 that is connected to the AUX1 pad. So the only way to use the AUX1 pad is to use SoftSerial.
If i get it to work it is still an option to use or not, telemetry is already working on UART3, I have it working on my other quad.

llambkin, 2016-08-11:
I just thought that if we are going to put some time into getting KISS ESC
telemetry working (which I think is attached to UART3) that we may loose a
UART.

basdelfos, 2016-08-11:
True, it is not really an issue to swap over to UART3 but i also think ESC telemetry is also connected to UART3.

basdelfos, 2016-08-11:
@borisbstyle 
Good point about to short the RX/TX pins when using telemetry on softserial, did not think of that. Had to do this on my old Naze F1 board too, forgot all about it. How did the smart thinkers at Flyduino do this. If they use softserial for telemetry than the AUX1 pad must be also connected to another pin on the F3. I will see if I can hunt down some circuit on the board.

fabio-cecamore, 2016-08-21:
Are there developments on this ?

borisbstyle, 2016-08-21:
No....not enough hardware information to make sure that it works.
AUX1 seemed to have some additional components connected to it.

sticksen, 2016-09-06:
@basdelfos in your opening post you´re saying that telemetry is working when the SmartPort is soldered to UART1 or UART3 TX pad. But where can I find these pads on my Kiss board?

basdelfos, 2016-09-06:
@sticksen,
UART1 are the TX/RX pads on top of the board near the USB connector. 
UART3 are the TX3/RX3 pads on the bottom of the board in the middle.
You only need to solder the telemetry wire to the TX or TX3 pad.

borisbstyle, 2016-09-13:
closing this issue as it is same as this one https://github.com/betaflight/betaflight/issues/673

closed 2016-09-13 by borisbstyle

basdelfos, 2016-10-01:
Well it is not actually the same als #673, this issue is about to use the standaard AUX1 pin for SmartPort telemetry for freeing a UART. Issue #673 is about adding ESC Telemetry. But they are related tho, if you want to use SBUS, SmartPort, OSD and ESC Telemetry, there is one UART short.

This issue seem to be also solved by the sharing one UART between SBUS (RX) and SmartPort (TX) i see coding for that in the master branch. This method also frees one UART. Don't know if the code for sharing UARTs is recently added, can't remember it being discussed the rcgroups thread in the past.

llambkin, 2016-10-02:
That would be awesome if that can be achieved. But I think the roadblock is that the UARTs cant be shared with different baud rates, I think that SBUS and S.Port are at different bauds

mikeller, 2016-10-09:
> I think that SBUS and S.Port are at different bauds

Yes, they are indeed, 100000 (SBUS) vs. 57600 (S.PORT). Thanks heaps for this, FrSky.

basdelfos, 2016-10-10:
Different baud rates, so sharing the same UART for SBUS and S.PORT is not possible. :-1:

sebest, 2017-03-24:
I think I got telemetry working on AUX1 using smartport on softserial1 (betaflight 3.1.6)

basdelfos, 2017-03-24:
@sebest, yes smartport telemetry works perfect on softserial with BF 3.1.6 and up.

sebest, 2017-03-24:
@basdelfos ,  thanks for confirming!
I wanted to spread the information as when I was looking for how to get telemetry with BF/KissFC/FrSky all the information online was saying that we have to use TX or TX3 and that AUX1 was not supported.

andrejpodzimek, 2017-07-17:
@sebest Pardon my ignorance, but how can telemetry work "on AUX1"? All howtos about SmartPort telemetry I've stumbled upon just say that you need both RX and TX to make it work, i.e., at least two pins. [Here's a SoftSerial configuration](http://fpvlab.com/forums/showthread.php?46100-New-FrSky-XSR-%96-3-8g-%96-buy-it-now&p=781540&viewfull=1#post781540) of that kind. And then there's [the well-known diode trick](https://github.com/betaflight/betaflight/wiki/Board---BLUEJAYF4#user-content-frsky-smartport) for BlueJayF4, again using **both** RX and TX (of a hardware UART this time). Because I need all the UARTs for SBUS, OSD and GPS, a magic method to make SmartPort work on one single pin (e.g. A13 or A14, both of which [might be free on my board](https://github.com/betaflight/betaflight/pull/2597#issuecomment-315662792)) would be more than great.

basdelfos, 2017-07-17:
@andrejpodzimek the links you refer to are mainly old information. In the meanwhile there BF firmware changed so that bi-directional softserial is possible. SmartPort is a bi-directional protocol, but it doesn't send and receive at the same time. Sending and receiving is done after each, that's why there is also only one wire at the receiver side.

I don't know which FC you are using, but depending layout/pins/configuration of the FC it should be possible to assign a softserial port to one of the free pins. 
But the best way to to find out if it works, just solder the smartport wire to A13 or A14 and configure the smartport resource to the pin you soldered the wire onto, e.g. in the cli `resource SERIAL_TX 11 A13`. Configure BF to use features SOFTSERIAL and TELEMETRY and set Softserial1 in the Ports tab to protocol Smartport. If it doesn't work, try A14 or another free pin.

andrejpodzimek, 2017-07-17:
@basdelfos Many thanks for the info. The CLI command you mention sheds more light on many of my RX/TX questions. I wasn't sure whether a SoftSerial would work with only the TX part configured, i.e., whether Betaflight would figure out what to do or just silently refuse to use a port that doesn't have an RX pin.

The board I'm using is a BlueJayF4 revision 3, [described here](https://github.com/betaflight/betaflight/wiki/Board---BLUEJAYF4#user-content-wiring-diagrams) (the last hexacopter-capable revision). [Here's my `resource list` output.](https://github.com/betaflight/betaflight/pull/2597#issuecomment-315662792) Earlier in that thread, someone wrote that the debug pad was connected to `B03` and usable for SoftSerial. But in `resource list` I see it assigned to `SPI_CS 3`, it's not `FREE`. That's somewhat confusing.

`A13` and `A14` are both `FREE`, but they are quite likely not connected to any pads. Soldering stuff directly onto the ARM processor doesn't look like a great idea to me.

Because I don't plan to use the SD card slot, it would be great to repurpose the pins that go to the SD card. (It would be easy to make a connector out of an old broken SD card.) But I have no idea which pins those are (`A15` and `D02` for sure, but there must be more pins) and whether they are suitable for SmartPort over SoftSerial.

basdelfos, 2017-07-17:
The pins described here are specific to other targets, every target has different pinlayout and probably doesn't work in your case.
If you don't use them, you could free the resource on motor pads 5 or 6 (B00/B01) and assign softserial to one of those pads and test is it will work. If you want to know which pins are connected to motor pads 7 and 8, you will have to ask the manufacturer or figure out yourself with a multimeter.

andrejpodzimek, 2017-07-22:
I’ve heard about the multimeter theory a number of times already, but I don’t think it’s practically feasible to figure this out using a multimeter. More often than not there’s quite a substantial resistance along the way between a pin and a pad. For the debug pad, for example, resistance toward STM32’s pins varies between 8 kiloohms and tens of megaohms. There is no obvious direct connection and pins with the lowest resistance towards the debug pad (8 and 64 kiloohms) are nowhere near B03 — if I’m guessing its location right, because there is no big square pad or anything that would reveal the chip’s orientation. So it seems that the statements about the debug pad on a BlueJayF4 connecting to B03 are either simply wrong or specific to one of the numerous clones of that board. (And presumably, SmartPort doesn’t work when softserial is on B03 and SmartPort connects to the debug pad.)

andrejpodzimek, 2017-07-24:
Sadly I can't use motor pads 5 and 6, because then I wouldn't be able to control motors 5 and 6.

As for the debug pad on BlueJayF4, SmartPort doesn't work on it. [I described the details here.](https://github.com/betaflight/betaflight/pull/2597#issuecomment-317303038)

It may be the case that the debug pad lacks a timer or other hardware needed for softserial. Then the only remaining option might be to find an unconnected pin that has the right hardware behind it, but I'm not sure it's safe to solder stuf directly onto those tiny pins