betaflight/betaflight #958: Sparky 2.0 accelerometer bug.

frandiazc, 2016-08-10:
In the gui, the model moves erratic when I move the FC and sometimes it freeze.

Tomcatt, 2016-08-14:
Mine always has a slight drift to it but I've ways see it in all the xFlight apps.  Are you able to upload a small video of it?

Tomcatt, 2016-08-16:
Forgot to leave a note to you all but I did a hover test and there's time where it would hold just fine in manual mode then I get a tiny jump.  Nothing crazy but it's concerning.  I'll play with the default PID's; that may fix that.

conkerkh, 2016-08-19:
If you use UART for MSP connection it's not erratic move but Packet errors. I reported this as an issue but seems that no one cares. USE VCP for your connection.

nathantsoi, 2016-08-19:
We care! Thx for reporting this and the other issue. There lots of things
to fix / review. I have a sparky 2, so I'll check this out.

BayaroX, 2016-09-11:
Is it right, that das is "only" a optical problem? The quad won`t show moves over yaw-axis and the other axis shown very slowly/freezing in configurator - but while flying this is not a problem?
So it`s possible to fly with sparky 2.0? Cause we have the same bug with our 3 sparky 2.0.
We use USB VCP to connect sparky 2.0 to computer.

closed 2016-12-19 by borisbstyle