betaflight/betaflight #959: [Bug report]STM32F4 LedStrip

Linjieqiang, 2016-08-10:
[image]
[image]
My flight control board used STM32F4 MCU.I use TIM1 channel1 to show led strip but it looks not working.Are there any bugs in STM32F4 LedStrip?I saw the code and found it.Maybe it was wrong?Only support TIM5?oh,I use betaflight firmware code in development branch.

Linjieqiang, 2016-08-10:
@blckmn Can the ledStrip work well on your fc board?Can you help me to fix it?Thanks.

HaseebNain, 2016-08-10:
I've been trying to get the Ledstrip working again and I also noticed that the IRQ Handler is never being entered. Even with TIM5 being generalized to WS2811_TIMER, when I try and create a specified IRQ handler the compiler ends up complaining about multiple instances of the IRQ Handler.

Linjieqiang, 2016-08-10:
@HaseebNain Yeah,I even tried to replace WS2811_TIMER to TIM5.But it didn't work.So the IRQ handler may be fixed,

HaseebNain, 2016-08-10:
I have gotten them working on an F4 but trying to port that code over to Betaflight hasn't been working as smoothly as I'd hope.

HaseebNain, 2016-08-10:
@borisbstyle Is the LEDSTRIP as is, working on any F4 Betaflight board?

borisbstyle, 2016-08-10:
I didnt hear any complaints. I personally never tested it

HaseebNain, 2016-08-10:
@blckmn It looks like the DMA IRQ handler is not being entered? I've tried setting up the interrupts using both the "dmaSetHandler" function and manualy setting up the interrupt using the NVIC structures in the   ws2811LedStripHardwareInit function.

I've flashed my variation of raceflight code and the strip seems to be working fine. I'm debugging using a toggling LED to see if the interrupt is ever called.

kc10kevin, 2016-08-11:
I have tried to set up the LEDs on the FuryF4 with no success.  Uses PA0, TIM5, Ch1.  Works in Raceflight so something not yet right in the code. The FuryF3 works flawlessly.

sambas, 2016-08-11:
can you try https://github.com/betaflight/betaflight/pull/973

HaseebNain, 2016-08-11:
@sambas I just compiled #973 with no problems. I can't believe I forgot to check the size of the buffer!

Unfortunately still nothing on the Oscilliscope and it still seems that the IRQ is not being entered. Were you able to get the strip working on the Colibri you were testing?

blckmn, 2016-08-13:
Check that the led strip is even getting an "update led" command from the scheduler.

I have a feeling the priority is not high enough and it never fires, thus never triggering the DMA sequence.

sambas, 2016-08-13:
This current master + patch for F4 and revo https://www.youtube.com/watch?v=U-Lemd7rHVk

mikeller, 2016-08-13:
@sambas: A bit off topic, but you can select multiple LEDs at the same time in the configurator by clicking / dragging over the LEDs you want to select.

kc10kevin, 2016-08-14:
Finally got around to testing this.  Works on the FuryF4 now!  Nicely done!  Thanks @sambas.

HaseebNain, 2016-08-15:
LEDs are working incredibly on the SPARKY2 as well, nicely done guys!

blckmn, 2016-08-16:
Working nicely here on bluejay too. I'll close this issue.

closed 2016-08-16 by blckmn