betaflight/betaflight #962: CLI "Allowed values" formatting

etracer65, 2016-08-10:
This cosmetic problem was reported in the forums a while back and had thought there was a pull request to fix it.  Maybe there was a regression?

http://www.rcgroups.com/forums/showpost.php?p=35115394&postcount=32531

In all the 3.0 versions the CLI output on OSX looks like this:

`# get dterm`
`dterm_lowpass_level = HIGHAllowed values: NORMAL, HIGH`
` `
`dterm_lowpass = 100Allowed range: 0 - 500`
` `
`dterm_notch_hz = 0Allowed range: 0 - 500`
` `
`dterm_notch_cutoff = 150Allowed range: 1 - 500`
` `
`dterm_setpoint_weight = 120Allowed range: 0 - 200`

It just looks like there's a missing line break before the "Allowed range" text.

borisbstyle, 2016-08-10:
I thought that was fixed. Is this the latest master?

KiteAnton, 2016-08-10:
Yes,  I fixed this previously in PR #600 but maybe the change got lost later somewhere. I can fix this again later tonight.

etracer65, 2016-08-10:
Yeah, I thought it was fixed also.  But it's still showing up in the latest builds.  The sample above is from 3.0RC6.

KiteAnton, 2016-08-10:
Ok, fixed it with #963 ,

borisbstyle, 2016-08-10:
Maybe it went into dev branch

etracer65, 2016-08-10:
I just tested and it looks good.

`# get dterm`

`dterm_lowpass_level = HIGH`
`Allowed values: NORMAL, HIGH`
` `
`dterm_lowpass = 100`
`Allowed range: 0 - 500`
` `
`dterm_notch_hz = 0`
`Allowed range: 0 - 500`
` `
`dterm_notch_cutoff = 150`
`Allowed range: 1 - 500`
` `
`dterm_setpoint_weight = 120`
`Allowed range: 0 - 200`

PR #963 (2016-08-10): Fix issue with newline for MAC users
Fixes #962
closed 2016-08-10 by borisbstyle