betaflight/betaflight #965: Target information

borisbstyle, 2016-08-10:
As the targetbase is getting larger and larger there is a need for a small readme file in each target directory what describes who is the board owner....what kind of board it is and where to find.
At the moment even I have no idea what some targets are.

Basically everybody who adds a target becomes a target owner. When things go wrong that person should fix the target if it is only target related.

digitalentity, 2016-08-10:
Agreed. I'll add README to #954

maxsymba, 2016-08-10:
Format of the info? I made XML file with all pins aasignment for stm32f3. What else?

Linjieqiang, 2016-08-10:
Agreed too.There are many targets……

mikeller, 2016-08-10:
@maxsymba: Markdown (https://guides.github.com/features/mastering-markdown/) is probably best, as GitHub can natively display it.

maxsymba, 2016-08-10:
@mikeller No no no... I mean other thing. 
What kind of information is necessary? Technical for developers or simply description of particular FC. Have anybody an example?

martinbudden, 2016-08-11:
@maxsymba , the following would be useful:

1. Description of FC for users, including link to user manual if available
2. Technical information for developers, including link to technical spec of FC if available
3. If applicable, link to website of producer/designer, eg link to http://www.seriouslypro.com/spracingf3evo for SPRACING EVO FC

MJ666, 2016-08-11:
For most initial targets ther is an file with information in the CF doc directory. This was removed in BF some time ago. For sure new target don't have this yet.

NightHawk32, 2016-08-28:
This is a good idea, so people also can contact target owners if changes will affect their target.
So the question is where to start this:
Adding a document to the corresponding target folder? Doing it like CF in a doc section? Using the BF wiki which already has sub-pages for each target?
Introducing a template for board descriptions is also a good idea, I think @nathantsoi already started this in the wiki. This would help to gain an overview about the different targets and their special features.

closed 2016-12-19 by borisbstyle