betaflight/betaflight #966: OSD not working on KISS FC Target

llambkin, 2016-08-10:
Hey Guys,

I have no idea when it stopped working, but it no longer displays anything provided by serial. The OSD powers up and shows the craft name, 0 volts and 0% throttle. It doesnt show the usual No Data Received when its connected the wrong way. I also cant get into the menu.

I can connect to it using serialpassthrough 0 and MWOSD, I can display various things up on the screen using the GUI, but when I connect the FC again, it still displays no serial data (voltage, throttle percentage etc.

I am using UART1 if that means anything, I cant get to UART 3, since its on the other side of the board I cant get to in this craft.

mikeller, 2016-08-10:
Just a thought, could this be caused by the OSD not being compatible with the recent changes to the MSP protocol?

llambkin, 2016-08-10:
Im doubtful, my REVO is also running RC6 and few issues there, but it is
working. MWOSD is a bit glitchy, but not surprised since all of the changes
lately.

Arakon, 2016-08-11:
Just tested with an SPRacingF3 clone, displaying battery voltage and mode just fine.

llambkin, 2016-08-11:
Yeah, I think it may be target related.

basdelfos, 2016-08-12:
Tested yesterday on a KISS FC with Betaflight and OSD on TX/RX pads, working without a issue.
But I have to say I'm using a custom OSD firmware that I'm currently working on. It is a totally rewritten version of MWOSD with only the basics for acro and freestyle flying. No need for a GUI also anymore, all settings are changable via the OSD menu. Layout it fixed other that top or bottom of the screen. I also made a KISS FW target of the OSD fw, it can show used mAh from the ESC telemetry in a nice percentage bar. 

I need to push it to github and make a nice website for it, didn't get the time for that yet. Will post the OSD fw in this thread for you to test.

basdelfos, 2016-08-12:
Here is the OSD custom firmware. It's called Beta-OSD at the moment. Need to get up with another name as there is now also a version that is compatible with the KISS FC propriety firmware.

Steps to use:
- Just flash the OSD firmware on the (micro) minimOSD.
- Connect RX/TX and 5V/GND and video signals.
- It initializes itself on first startup and installs the font automatically.

Accessing the OSD menu is the same a with MWOSD: Mid-throttle, yaw-right, pitch-up at the same time when disarmed.

PID turning and Rate tuning is compatible with BF 3.0 or higher.

Configuration as Vbat min/max/warning and RSSI is all read from the configuration in Betaflight. For RSSI % to get working you need to pass the RSSI signal back on a spare channel from your Taranis to Betaflight. See Google for instructions. But you can also just disable RSSI in the OSD menu, there is no use for us miniquad races/freestylers anyway.

Video mode is PAL (if you need NTSC let me know).

If you find anything, please let me know.

[Beta-OSD 2.0.6 BTFL.hex.zip](https://github.com/betaflight/betaflight/files/414730/Beta-OSD.2.0.6.BTFL.hex.zip)

llambkin, 2016-08-12:
Awesome, it will be fun to test out new OSD code again. I helped shiki out getting MWOSD working for KISSFC. I will stick it on some time over the weekend. Unfortunately I am not running KISS FC on my craft with KISS ESCs any more, so I can't test the ESC telem. I am having too much fun with my F4 REVO.

llambkin, 2016-08-12:
You could call it DelfOSD ;-)

basdelfos, 2016-08-12:
Yeah something like that my fly mates also mentioned, BasOSD or OSDelfos. But that's a little selfish in my opinion ;-)

llambkin, 2016-08-14:
Hi, I just tried uploading it via xloader, is ther an .ino file you can send me? It keeps freezing for me....

llambkin, 2016-08-14:
All good, forget my last. I had the RX/TX mixed :D

llambkin, 2016-08-14:
OK, a few of things,
1: Is it going to have an option for mAh? I am about to connect a current sensor, and it must have mAh reading.

2: Is there some sort of dead zone you can put on it, I am finding it jumps a couple of options when I flick the TX stick over, maybe it could be the superexpo coming out of the FC. @borisbstyle or @mikeller , is it possible that the rates are affecting the OSD? If so, is there a way to switch off superexpo or high rates for when the craft is disarmed? (I just thought of this, probably not possible due to the configurator needing it for the rates testing in the GUI)

3: The cursor stops at a few positions around where TPA is set, maybe some cleanup needed there?

4: Is there a way of uploading new FW to the OSD without having to resolder, maybe I need to flash a new bootloader so the FC can trigger a reset to the atmega, then new FW can be uploaded using xloader, and or possibly the Betaflight Configurator in the future.

Maybe we need to set up a GitHub just for testing this OSD.....

llambkin, 2016-08-14:
@basdelfos, Scratch my 1)..... I have found your GitHub for Beta-Osd :D. Looking into configuring now, Maybe you need to put a mAh yes/no option in the config screen on the OSD.

mikeller, 2016-08-14:
Re 2.): At least for the configurator, what is actually read over MSP are the RC channel values as received from the RX. All rates and expo is applied in the configurator for the model display. I would assume that OSD is probably using the same command to read RC channel values.

llambkin, 2016-08-14:
My problem is that the OSD jumps about when going through the menu (on both MWO, so it

closed 2016-08-19 by borisbstyle

Markuzzzz, 2017-06-11:
Hey Guys, I am testing the MW Osd for Kiss. I have a strange error going on. When ever I use level/acro mode selection using a switch on my TX the osd disappears entirely. Not matter on which Aux I set level/acro mode in the Kiss Gui. Anybody having the same problem? Shouldnt it react only on AUX4?

Markuzzzz, 2017-06-11:
Tried 1.6.8 and the newest beta of 1.7 for Kiss from github.

basdelfos, 2017-06-11:
Do you have Betaflight running on you KISS FC?
If yes, than you will need to flash MWOSD to use Betaflight not KISS.

Btw; i think you are requesting support in the wrong project. For help on MWOSD you can better ask for support in their github project. You will probably get the right answer quicker.

Markuzzzz, 2017-06-12:
I think you're right, wrong thread 😊