betaflight/betaflight #969: fly away BF 2.9.0

Philippe-France, 2016-08-10:
Hi

I got a fly away with BF 2.9.0 on BeeRotor F3 flight controller flashed with betaflight_2.9.0_SPRACINGF3.hex.
That was the 5th lipo after maiden.

You can watch the fpv record here https://www.youtube.com/watch?v=5kjbjx3p68w

Power up at 3S lipo plug was all right, as soon as I armed the quad fly away  while throtte was kept at minimum and strange buzzer alarm occured, I succeed to disarm but even here the FC continued to control motor to keep attitude.

Air mode disabled. The most strange think is the FC controlling motors after disarm action.

Back to the bench all was ok.

Flashed back to BF 2.7.1 and ran 6 lipo, all ok

Any idea ?
Forget me if not the right place to post such issue, but according to me is really a SW issue as FC was controlling the attitude after disarm

thank

borisbstyle, 2016-08-10:
Check your failsafe first before flying!

This sounds like failsafe from rx

Philippe-France, 2016-08-10:
Could be yes, may be a bad init of the RX
Failsafe was set and tested before and after on bench with quad attached: worked fine

borisbstyle, 2016-08-19:
Sorry this looks like power loss. You can clearly see osd going off

closed 2016-08-19 by borisbstyle