betaflight/betaflight #971: Betaflight Lockup when using Multishot

antbarney, 2016-08-11:
Having a issue with Betaflight 2.9.1 (3.0.0-RC6 also does it). If using Multishot on a flip or roll some times it appears the FC will lockup and reboot (based on what I see in the logs, I can also hear the beeper tone for start up.) this causes the quad to flip as 2 motors are normally going full throttle to start the roll at this time. For fun since I had a few I have done this using 2 SPRF3 Clones and does it using either of them. I also tested switching everything back to OS125 and everything works as expected. Not sure if it could be the ESC's or not but I have a set of Blheli_S ESC's coming soon that I can also try with.

Specs:
SPRacingF3 (Clone)
RotorGeeks RG20A+ ESC's flashed with BlHeli 14.6 Multishot.
EMAX RS2205 2300kv
220 Size X frame.
KingKong 5x4x3 props.

CLI Dump and a Blackbox log of a test flight to get it to lockup and flip.
https://1drv.ms/f/s!Avk4-2mpBlgfiOxmj-waWXLQ5k_lZg

QuadroBro, 2016-08-17:
I'd check your solder joints and connections... Given that it only happens on flips and rolls it sounds like your losing power briefly. What's your CPU usage when connected to BF?

antbarney, 2016-08-17:
I had thought the same before posting so I went thru and checked all the solder joints on at least one of the boards as well as the build. The CPU usage is sitting at about 16% so I don't think its CPU related. The biggest thing that was weird to me was the fact that it only does it if using MultiShot, if I Switch back to Oneshot125 no issues. I think it may just be cheap boards so I have some X-Racer F303 v3.1 on the way and will see if the issue persists after switching to one of those. But I do find it odd that 2 different SPRF3 boards do the same thing in this setup.

antbarney, 2016-08-17:
Closing this down, no issues so far with this build once I swapped to a Xracer F303

closed 2016-08-17 by antbarney