betaflight/betaflight #972: BLHeli Passthrough on XRacer F303 V2 and V3.1 when anything is connected to UART1

drkavnger99, 2016-08-11:
It seems if you have UART1 connected to anything (in my case smartport from X4R-SB) even when disabled in configurator will give random connectivity to BLHeli.
BLheli 16.2.14.6.0.1 Betaflight 3.0 RC6 and Betaflight 2.8.1 display the same issue.

Arakon, 2016-08-11:
Well, yeah.. you have a device with resistors and capacitors etc connected to the lines that are transferring data.. it's fairly normal that you get interference from that.
UART 1 is the same port the USB port uses.

lostcontrol, 2016-08-11:
@drkavnger99 Just soldered my XSR SmartPort to that UART yesterday evening :disappointed: I have to check if I also have issues now with BlHeli.

@Arakon Then MSP should be also affected e.g. when connected with the configurator. That does not seem to be the case, at least, I did not notice anything strange yesterday. The connection to ßF configurator was rock-stable.

19Gerhard85, 2016-08-11:
Following as I'm about to rebuild my quad that way (Telemetry on UART1, OSD on UART2).

lostcontrol, 2016-08-11:
Checking [Cleanflight's documentation](https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md), I see that:
> - Smartport telemetry cannot be shared with MSP.

So UART1 seems to be a bad choice for Smartport anyway. Don't know why fpvmodel put this on their picture. Better use UART2 or Softserial for Smartport. I will solder mine to UART2 since it's free.

19Gerhard85, 2016-08-11:
Don't know why it shouldn't be shared with MSP. The Rx will only receive data from the FC, so it shouldn't be a problem at all. 
Can anybody clarify this?!

lostcontrol, 2016-08-16:
So I finished my build and did some tests with BlHeli passthrough with my Smartport connected to UART1. No problem so far!? Telemetry is working, I can connect to all ESC, read the settings, change them, etc. I did not try to upload new firmware to the ESC though. I only have TX soldered on UART1. Not yet flew with this build but I will keep my Smartport on UART1 for now and see how it behaves.

Also having MSP and Smartport telemetry configured on this port does not seem to be an issue (as mentioned in the Cleanflight documentation).

borisbstyle, 2016-09-07:
Closing this as it is not a real issue

closed 2016-09-07 by borisbstyle