betaflight/betaflight #983: Custom motor mix not working

UltiFix, 2016-08-12:
So I soldered my esc wires to my FC in the wrong order - The numbers in red are what motor output control that actual motor. So motor 1 on the betaflight motor tab controls motor 3...

So I made a custom motor mix, and set my copter to USE the custom motor mix.
mmix 2 1 -1 1 -1
mmix 3 1 -1 -1 1
mmix 0 1 1 1 1
mmix 1 1 1 -1 -1

But no matter what I do motor 1 on the betaflight motor tab keeps controlling motor 3.

Totally lost.... I also watched this video.

RipperDrone, 2016-08-12:
Think you need to set the mix type to custom as well - have you done so?

If it's still Set to e.g. QUAD, the mmix switched Don't become effective

UltiFix, 2016-08-12:
Yep, mixer custom

RipperDrone, 2016-08-12:
Another thing: In earlier versions the sequence of mmix commands always had to start with Channel number 0 and then line by line count up, like

mmix 0 ...
mmix 1 ...

In One of my settings I even had to start with mmix 0 ... despite using Channel numbers 2+ only.

Worth a try?

UltiFix, 2016-08-12:
Here is a exact copy of the commands I did... Latest betaflight on naze32 rev6

mmix load custom

mmix 0 1 1 1 1
mmix 1 1 1 -1 -1
mmix 2 1 -1 1 -1
mmix 3 1 -1 -1 1

mixer custom

mmix

save

RipperDrone, 2016-08-12:
This it what works on my lowrider ZMR250, SPR3:

mixer CUSTOM
mmix reset
mmix 0 1.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000
mmix 2 1.000 -1.000 0.772 -1.000
mmix 3 1.000 -1.000 -0.772 1.000
mmix 4 1.000 1.000 0.772 1.000
mmix 5 1.000 1.000 -0.772 -1.000
smix reset

UltiFix, 2016-08-12:
Tried what you did (just replaced the mmix 0 1 etc commands with mine)

And it didnt work D:

RipperDrone, 2016-08-12:
changed the mixer CUSTOM command as well and put it in front?

RipperDrone, 2016-08-12:
try copy-paste the whole following lines into your CLI. This should be your desired settings:

mixer CUSTOM
mmix reset
mmix 0 1.000 1.000 1.000 1.000
mmix 1 1.000 1.000 -1.000 -1.000
mmix 2 1.000 -1.000 1.000 -1.000
mmix 3 1.000 -1.000 -1.000 1.000
smix reset
save

UltiFix, 2016-08-12:
Copy pasted and it didnt work.... In Betaflight turning up motor output 1 spins motor 3 (standard BF layout)

UltiFix, 2016-08-12:
Somebody on reddit told me this

"
The motors tab is going to control the motors in the order you wired them. Motor mix doesn't change that. You should be ok to fly. I had the exact same hesitation when I did mine, but it works.
"

RipperDrone, 2016-08-12:
Try checking correct hard wiring to Naze motor outputs, then play with mmix custom settings by shifting the 1 and -1 figures around and see if motor mapping changes

RipperDrone, 2016-08-12:
Maybe reddit is correct and you should give it a try in real flight

UltiFix, 2016-08-12:
I ended up manually rewiring it - and it flies :D

So I guess the mixes where working correctly, they just dont effect the motor tab.

closed 2016-08-12 by UltiFix