betaflight/betaflight #985: Content Issue

blckmn, 2016-08-12:


blckmn, 2016-08-12:
[image]
[image]

closed 2016-08-12 by blckmn

blckmn, 2016-08-12:
[image]
[image]
[image]
[image]

blckmn, 2016-08-12:
[image]

mikeller, 2016-08-12:
Nice pictures. Now I want one. ;-)