betaflight/betaflight #986: barometer BMP280 show incorrect values in v3.0 RC6

siamect, 2016-08-12:
The barometer BMP280 show incorrect values in v3.0 RC6. It is showing the derivative value so it returns to zero when the vertical movement stops even if the absolute altitude has changed.
Sometimes the altitude changes to strange values like -2550m and sometimes just a meter or two.
I can verify this doesn't happen in v2.91

siamect, 2016-08-22:
...pretty weird... the v3.0 RC6 I loaded the hex file from the releases and it had the problem described... The v3.0 RC8 I built by my self and it works. Self-repairing code??! I also seems to behave better when I use throttle in baro mode, but I need to test that more...

nathantsoi, 2016-09-15:
have you upgraded to 3.0 final, @siamect. is this still broken?

siamect, 2016-09-16:
Thanks Nathan, no not yet...

Serbitor, 2016-10-15:
I can confirm that this is still broken in 3.0 final

Serbitor, 2016-10-15:
Further testing on my spracingf3mini:
3.0.0 build on my machine; barometer works as expected
3.0.0 installed from "Load Firmware [Online]" the barometer doesn't work.

Tested this by watching the barometer chart in configurator while lifting the quad from the desk to 2m higher.
On my build the chart shows two different levels approx 2m apart.
On the official build the value drifts back towards 0 after moving it up or down.
I suspect there is something wrong with the official image, any chance of a re-release for this board?

@siamect what board were you testing with?

siamect, 2016-10-17:
@Serbitor 
Sorry for my late answer. The board is from Bangood product id is 1021113.  
"Mini F3 Flight Controller 6 DOF/10 DOF for Multicopter Racing" it says and it is a SP3 racing clone (not the SP3 mini so it runs the same software as the big SP3 racing board) and it worked well and I was happy with it but then I fried it. The V-bat cable got pierced by one of the solder pads that was sharp and that was the end of life... a sad day and it was not even funny. No smoke no fire nothing... just a silent death :)  
Now I got a new controller, yet another clone but from Hobby King (571000088-0) and it is a "Mini CC3D Revolution 32bit F4 Based Flight Controller". It is really tiny and it has a different baro sensor (MS5611)  that works fine with Betaflight 3.0.0. (REVO) both the ready built binary and the one you get if you compile it by yourself.

AndersHoglund, 2016-10-21:
Strange if this should be a gcc4 vs gcc5 issue. Certainly seem to appear that way...
Any difference with the Jenkins 3.0.1 release buiild using GCC 5.4.1 on http://andwho.sytes.net:8080/job/BorisB_BetaFlight/618/ ?

Serbitor, 2016-10-21:
@AndersHoglund thanks for linking those builds. The GCC 5.4.1 build does still have the problem with baro reading drifting back to 0

AndersHoglund, 2016-10-21:
This is sooo weird. And I do not have any FC with baro for a debug session...
What happens when/if you update your gcc installation?

AndersHoglund, 2016-10-22:
If you have the same result ( baro "OK" ) with both 4.9.3 and 5.4.1 then the problem lays somewhere else.
For the record, Jenkins uses this gcc: 

    arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

A bit odd that two gcc 5.4.1 release versions are slightly different.....

Serbitor, 2016-10-22:
I just downloaded the 2016-q2 toolchain from launchpad and when building with this version, the problem returns. Looks like an update to the 2016-q3 toolchain would be a good idea.

AndersHoglund, 2016-10-22:
Odd. Should only the 5.4.1 2016q2 introduce a derivative baro measurement value, as opposed to absolute value with 4.93 and 5.4.1q2. Really odd.

Anyway, I am preparing a PR for a SDK update. But there are side effects, binary images are slightly bigger. This causes CC3D and CC3D_OPBL targets to fail:

    $ tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc --version
    arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
    .......
    LD betaflight_CC3D.elf
    /home/andho/projects/cleanflight/andwho/betaflight/tools/gcc-arm-none-eabi-5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: obj/main/betaflight_CC3D.elf section `._user_heap_stack' will not fit in region `RAM'
    /home/andho/projects/cleanflight/andwho/betaflight/tools/gcc-arm-none-eabi-5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 88 bytes
    ......
    LD betaflight_CC3D_OPBL.elf
    /home/andho/projects/cleanflight/andwho/betaflight/tools/gcc-arm-none-eabi-5_4-2016q3/bin/./lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: obj/main/betaflight_CC3D_OPBL.elf section `.data' will not fit in region `FLASH'
    /home/andho/projects/cleanflight/andwho/betaflight/tools/gcc-arm-none-eabi-5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 140 bytes
Darn....

Serbitor, 2016-10-22:
I agree that is is a strange one, and quite annoying that it breaks the smaller targets.

AndersHoglund, 2016-10-22:
Would be nice if some one else also could confirm those baro readout findings.

Serbitor, 2016-11-01:
@AndersHoglund i feel like an idiot right now.
I managed to get out for a test flight and found that my build was no better than the official one. Toolchain is not the problem you are right to be suspicious.
If I try 2.9.1 with accelerometer disabled the baro works. If I do the same on 3.0.1 the baro sensor constantly returns 0.

I ran a git bisect between 1da7883 (current master) and 0ef6c3d (2.9.1 release).

For each build I did a full chip erase and ran these cli commands:
```
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = AUTO
save
```

I found that e9a46c6ebbc32094a3565ea1498f6ca7162c9489 seems to introduce the problem.
Having looked at the PR I don't understand how that has broken the bmp280 baro on my board.

Apologies for calling wolf on the toolchain

AndersHoglund, 2016-11-01:
OK, good you are digging deeper into this. 
Just to check, exactly what FC and baro is this? You mention both bmp280 and MS5611. What baro type does the cli "status" command report?

Serbitor, 2016-11-01:
My FC is SpracingF3Mini which has BMP280 baro. Behavior is the same on two 
separate FC.

On Tuesday, 1 November 2016 02:13:14 GMT Anders H�glund wrote:
> OK, good you are digging deeper into this.
> Just to check, exactly what FC and baro is this? You mention both bmp280 and
> MS5611. What baro type does the cli "status" command report?

micmax79, 2016-11-05:
I'm running a F3 Racing Deluxe "clone" from Banggood (Product ID: 1060619) with MS5611 pressure sensor. I'm also getting strange Baro readings. If I move my quad up and down, I can see the height difference in the sensors tab. As soon as I keep the quad in the same position, the readings start drifting in the negative direction. I'm quite sure that the drift is not a temperature drift as the F3 board was powered for nearly an hour. Furthermore, I get one I2C error which is related to the baro sensor. Maybe the clone board is defective or it's really a bug in the current Betaflight version. I'm running 3.0.1.

[image]

Serbitor, 2016-11-05:
@micmax79 can you disable the accelerometer and test the barometer again please

micmax79, 2016-11-05:
@Serbitor If only Baro is enabled, the height is **always** zero. I2C Error is still 1.

closed 2016-12-19 by borisbstyle