betaflight/betaflight #989: OSD: Boot Sequence ... OSD disapear after cam has powered up

peisel, 2016-08-12:
Hi,
i connect the power than FC boots and i see the OSD. The cam has a little delay to boot (RunCam Eagle) so when the cam sends the signal after OSD its overlays the OSD Display. When i rebooting the FC over CLI and leave it powered via lipo ... it shows correct the cam and osd. Timing Issues?

Issue:
https://www.youtube.com/watch?v=ZcdxWRsIXpk

Arakon, 2016-08-12:
The FC has no control over OSD display being active or not. If the overlay disappears, you either have a wrong NTSC/PAL setting or dirty power.

peisel, 2016-08-12:
Sure? Its a Omnibus FC with configuration for OSD over Betaflight GUI direct connected to STM

nathantsoi, 2016-08-13:
thanks for the comment @peisel. i watched your video and I will test. could you please try re-setting the layout to see if it works? maybe it's related to the layout issue you opened? also, what type of camera do you have pal or ntsc?

peisel, 2016-08-13:
Didn't found a option to reset the Layout ... this is what im looking for on the other one. The cam is a new RunCam Eagle (CMOS) PAL will try also a Swift later if it is a issue from the cam ... that she is sending to late a video signal after power. Had read about these issue with older firmwares of minimOSD ... i think they solved there also a issue that is similar there ...

SnappyFPV, 2016-08-13:
Ok, found the issue. AUTO Mode hasn't work with the eagle - because of the late picture start after power - default mode on AUTO is propably NTSC ... Cam was set to PAL so after set to NTSC on Cam it works also in AUTO... when i set PAL on OSD and PAL on Cam it works too. I didn't know if its possible to fix AUTO Mode also for cams that send picture later ...?

jflyper, 2016-08-14:
@nathantsoi Take a look at the AUTOCAM code for MWOSD (current master). It periodically  (along with stall detection) checks changes in NTSC/PAL bits of STATUS register, and switches to whatever detected. It's only a few lines of code.

syrekszymon, 2016-09-11:
Same issue on racebase (integrated osd) on 3.0 final firmware. OSD doesn't showin up or 
completely black screen. Revert to 2.9.1 and osd work fine but with wrong charset.

nathantsoi, 2016-09-15:
cool, that would be awesome @jflyper. any chance you want to submit a pr?

closed 2016-12-20 by borisbstyle

jflyper, 2016-12-20:
PR submitted: #1639