betaflight/betaflight #994: No Yaw Control

digitalomni, 2016-08-13:
I upgraded my SP Racing F3 FC to BF3.0 REV 6 yesterday. I also updated the configurator to 1.7.2. While setting up the quad in BF, I noticed that the virtual quad displayed was not displaying YAW control. Pitch and Roll was working fine. I thought it might be a bug so I continued on to a test flight.  Sure enough no YAW control and it actually started slowly turning counter clockwise when in the air.

I have a backup quad (exact build) tried the same 3.0 flash and got the same results.

So I flashed back to 2.9.1 which I had come from with everything working great, restored my config file, checked all my settings and again still no YAW.  So now I am thinking could it be the new 1.7.2 configurator???  Planing on reverting back to 1.7.1 if I can figure out how.

Anyone have any ideas?

Thanks

mikeller, 2016-08-13:
There was a bug in the configurator that set the yaw rate to 0. But that should be fixed in 1.7.2.

Can you go into the CLI, and check the output of `dump rates` to see if `yaw_rate` or `rc_rate_yaw` are 0. (And if so, set them to 70 or 100 respectively.)

digitalomni, 2016-08-13:
Thank you Mikeller,

Sure enough YAW RC RATE is set to 0 and I am using 1.7.2

mikeller, 2016-08-13:
The reset could have happened when you were using an older version. Can you set a value in the GUI, save, and check in CLI if the right value gets saved?

RipperDrone, 2016-08-14:
Seen same thing here. After upgrade and reloading old parameters, yaw rate was set to 0. Setting back to e.g. 1 fixed it. Save/restore from that point works w/o any issues.

mikeller, 2016-08-14:
As said above, there was a bug that caused this in a past version of the configurator. So unless a reset happens going forward from now on, this is fixed.

digitalomni, 2016-08-14:
I am going to check one more thing today. After resetting my YAW values in CLI, I did a flight test and it worked. But then I changed profiles to see if the problem was global or per profile. When changing to profile 2 the problem was back. So I changed it back to profile 1 and still had the same problem..  So I went back into BF and my YAW values had seemed to be reset back to 0. Could that be possible?

RipperDrone, 2016-08-14:
Maybe you only reset profile to profile 1, but not rateprofile? Yaw rate is part of rateprofile imo.

digitalomni, 2016-08-14:
Little confused, the new 3.0 GUI now supports 3 different Rateprofiles per profile.  When I go to CLI and perform Dump Rate it comes back with the rates for Rateprofile 0 and not 1, 2 or 3 defined in the drop down in the GUI.  And I'm sorry if this is a dumb question,  pretty new to all of this and I appreciate your patience.

borisbstyle, 2016-08-14:
That was always like that but never shown in the gui.
Almost nobody knew that

digitalomni, 2016-08-14:
I think it will be super useful for tuning. I guess I forgot to ask my question clearly above.. What is Rateprofile 0 in the GUI. Does Rateprofile 0 = Rateprofile 1?

RipperDrone, 2016-08-14:
Due to memory limitation reasons, profiles have been shortened from formerly 3 to now 2. In contrast, 3 rateprofiles have been kept. 

Both numbering schemes in CLI are zero based, e.g.

profile 0 or 1
rateprofile 0, 1, 2

You need to switch them separately.

mikeller, 2016-08-14:
@digitalomni 

Yes, CLI rateprofile 0 is the same as GUI Rateprofile 1, and so on.
The same goes for CLI profile 0 being the same as GUI Profile 1.

This is a bit annoying, but in a way 0 based is right for the CLI (on the flight controller), whereas 1 based is right for a GUI for less technical users, so no good way to do it right for every case (at least that I know of).

digitalomni, 2016-08-14:
Thank you all

closed 2016-08-19 by borisbstyle