betaflight/betaflight #1003: BF 3.0.0 RC7 - Baro Mode: Throttle Pumping

RipperDrone, 2016-08-14:
Baro Mode worked well after some PID tuning & adjustments in the code until BF2.9. From BF 3, there seems to be changes in Baro Mode which force my Armattan X155 / SPRF3 to 'pump' altitude up/down periodically (approx. 2 pumps per second). After trying to rgain control by several throttle stick up/down movements, it seems to eventually calm down.

Could be PID off target again? I am using your defaults now.

OR has there been any change in the Baro Mode PID loop / parameters in general?

I know Baro is not your primary focus, but for starting / landing I'd like to have the Baro mode working at least to an extent that I don't need to be scared... ;-)

alsop, 2016-08-14:
I am experiencing the same behavior with a Vortex 250 using 2.3.5 (the version it came with).  Adjusting the Alt. P value from 50 down to 10 does not make any difference.  The "sensor" display in Betaflight Configurator shows the barometer to be working correctly, although it does drift a bit (+/- 0.5 meters).  So I am guessing that the PID values displayed in Configurator are not the ones actually being used when BARO mode is turned on.  As delivered, the Vortex configuration had disabled the Barometer, so perhaps this is a well-known problem?

RipperDrone, 2016-08-20:
Just completed maiden of my new build QAV. Exactly same isseu with stock ALT/VEL PID. Pumps / oscillates altitude / thrust when switching back from acro into self-level with MAG & BARO (for landing).

borisbstyle, 2016-08-20:
@RipperDrone 
Baro // Mag is not the main priority of betaflight project. Those functions are better maintained in iNav and Cleanflight project.
In betaflight it is really best effort support. I wil park this for 3.1.

RipperDrone, 2016-08-21:
Acknowledged. Not asking for a butter smooth hold altitude here, just very basic alt hold is enough. In the Small group where I'm flying, everybody finds this mode helpful when landing. Excessive alt jumps can then crash the copter.

borisbstyle, 2016-09-07:
Can you try RC13 just to verify if this had to do with wrong baro detection? Reopen issue if it still persists

closed 2016-09-07 by borisbstyle

RipperDrone, 2016-09-07:
worked nice today however I noticed later that I had my ALT and VEL settings down to 30. No more pumping, not a single bit. Maybe you want to default the ALT / VEL to those lower values in the next rev?

Anyway, thank you for notifying me about the BARO detected wrong fix!

As you said, most of the time I'm using Acro / Air modes. BARO only sometimes to check things...

alsop, 2016-10-18:
Yes, the performance has smoothed out nicely.

-John

From: borisbstyle 
Sent: Wednesday, September 07, 2016 8:24 AM
To: betaflight/betaflight 
Cc: alsop ; Comment 
Subject: Re: [betaflight/betaflight] BF 3.0.0 RC7 - Baro Mode: Throttle Pumping (#1003)

Can you try RC13 just to verify if this had to do with wrong baro detection? Reopen issue if it still persists

Jerm357, 2017-01-26:
I'm having this same problem on 3.0.1 and 3.10 on my SPRacingF3 flight controller. I came from BF 2.6 and it was working pretty good but as soon as I upgraded to 3.0.1 and up I get this same Throttle Pumping as described above. I don't get any changes changing ALT or VEL P  but changing VEL D does have some kind of effect. Everything else is working great for me so it would be awesome to get my Baro working back like it was. Any ideas what this could be?

alsop, 2017-01-27:
Hi Jerm357.  Thanks for CCing me on your message.  I am still running 2.xx versions and have also found that some of the settings have no observable effect.  One thing that makes this a little more livable for me is to make sure my vertical speed is very near zero when I engage altitude hold.  But if I try to climb much, it takes off like a rocket – and then nearly stops the props entirely.  This is exciting visually, but not useful.

Jerm357, 2017-01-30:
Any way to open back up this issue so we can get this looked at again?

MrSmith2017, 2017-02-08:
Yes, would be nice when this can be fixed.

wogfun, 2017-02-15:
Experiencing same issue as described above on latest spf3 fc.

STOLerant, 2017-03-31:
I'm having the same issue, was there ever a solution?

andrejpodzimek, 2017-08-07:
The same issue still persists. I’ve just seen severe throttle pulses (with a frequency of ~2 per second) with the latest 3.2-to-be master build, [with this aircraft](https://www.rcgroups.com/forums/showthread.php?2593106-BlueJayF4-STM32F4-based-FC/page68#post38017641). Although my barometer is now working, thanks to the great advice from experts in the community, I can’t really use it, because my hex (well quads also have a short name, so why not hex) just jumps around the target altitude in a crazy way. I surrounded the baro with foam to rule out interference with propeller sound, but nope, it’s still **bad**, to say the least.

noviuk, 2017-10-20:
Betaflight 3.2.2
SP F3 clone
Had the same issue. Solved by changing P to 0, and increasing I to 15 with D to something like 10.
1st tests show that there is no more oscilation.

Glowhead, 2017-10-25:
I have never had a problem with the modified F3 versions pumping the throttle. With two SP Racing F3 boards only one has the pumping throttle issue. The modification:
ALT PID 50;0;0 and VEL PID 50;5;30 helped a lot! The altitude does float between the widest ranges of all four models but it works. I always land baro off. Not problems landing.

SpyQuad, 2018-08-21:
+1 for me  for Altitude Hold (Baro) PID settings ALT PID 50;0;0 and VEL PID 50;5;30. I have built a QAV250 quad with a "HobbyKing" Ghost Edition frame and MT2204 2300KV motors driven by KingKong 12A ESCs and an F1 class Illuminati 32 (Naze32 clone) Flight Controller with built in Baro (MS5611). I have experimented a lot with CleanFlight (different versions) but although Acro mode was working steadily, Baro mode never worked, or I could'nt make it work.
Then I switched to BetaFlight and I saw improvement in the quad's behavior from the start. I flashed BetaFlight 3.2.5, although some suggest not to try past the 3.0.1 version for F1 class FCs. In my case all went well and the quad's behavior was adequate from the beginning.
However in Baro mode I experienced the "either going up or going down" syndrome with no real "altitude hold". Then I saw this thread and tried VEL PID 50;5;30 as suggested, and voila !
Altitude hold works within let's say 0,5 meters. There is enough "dead zone" around the initial throttle stick position, so you can find the neutral position easily. When you move the stick past the dead zone the quad slowly begins to ascend or descent as expected. When you re-enter the throttle stick's neutral zone the quad again holds its altitude.
I should also mention that I have set:
set alt_hold_fast_change = OFF     (max alt change = +/- 0.5 m/s)
Now I just have some oscillations in Angle mode and I am trying with the motion PID settings to also overcome this. Any suggestions ?

MR444, 2018-08-31:
Baro mode was removed in 3.5.0. I started a feature request.
https://github.com/betaflight/betaflight/issues/6675