betaflight/betaflight #1004: Question. RC throttle dependecy on on ESC min_ and MAx

AndersHoglund, 2016-08-14:


AndersHoglund, 2016-08-15:
This is soo weird. I don't dare to touch this code, simply do not understand the thoughts behind it. 
[image]

Where are rxrange? Should be there really...

    # rxrange
    rxrange 0 1000 2000
    rxrange 1 1000 2000
    rxrange 2 1000 2000
    rxrange 3 1000 2000

borisbstyle, 2016-08-15:
I think rx range is missing and needs to be added.

But the entire RC code is just one big confusion and needs a heavy rework!
The plan is to do that after all priorities are finished.

Rc commands and stuff is being manipulated in so much places at the moment
that it is really a pain to maintain.

I really want to centralise that.

AndersHoglund, 2016-08-15:
OK, thnx. Not only me then.....
This only work (-ish) because the numerical values of the different factors just happens to be close to 1000 and 2000. Pure luck, not engineering.

closed 2016-08-15 by AndersHoglund