betaflight/betaflight #1006: Smartport Telemetry on UART2 not working

friszky, 2016-08-15:
Using Betaflight 3.0.0 RC7.

No telemetry is being sent to Taranis X9D+, other than what the XSR detects on its own (A2, SWR, etc.) Looking to get important stuff (VFAS, Alt, that sort of thing.)

Board is SPRacingF3 Mini, telemetry should be being broadcast via Frsky XSR, Smartport wire is soldered to UART2 Tx. Port configuration:

[image]

telemetry_inversion=ON, CLI dump is as follows:

# version
# BetaFlight/SPRACINGF3MINI 3.0.0 Aug 12 2016 / 10:50:30 (9ee9987)

# name
name TANGO

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature TELEMETRY
feature BLACKBOX
feature AIRMODE
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
[... 353 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75

# rateprofile
rateprofile 0
set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

#

borisbstyle, 2016-08-15:
Smartport works on all other boards so it should work on this one too.

Please recheck your connections

xeniter, 2016-08-16:
have similar issue, use smart port on uart 3(flex port) with cc3d

with stable release (Betaflight 2.9.1) it worked again with exact same config

read you change some io stuff in release notes, also smart port stuff?

yours,
xeniter

ichbtm, 2016-08-17:
same here with a piko blx fc and smartport on uart2  : 
taranis receive RX generated sensors (A2, SWR,...) but no VFAS.

friszky, 2016-08-18:
boris, you rock. I don't know why, but for some reason, after a series of arming/disarming while discovering sensors, everything's all good. To those having issues: try discovering while armed. try discovering while disarmed. try giving the board a power cycle between attempts. I know this isn't particularly helpful, but *some* combination works.

slawek89, 2016-08-19:
I have the same problem with CC3D on different software version. S-port from XSR is conected do UART3. (In ver 3.0.0 there is doubled UART 1).

borisbstyle, 2016-08-19:
I tested this and it simply works on all the targets I tried. Must be config issue

closed 2016-08-19 by borisbstyle

friszky, 2016-08-19:
Slawek, try all kinds of combinations of armed/disarmed and power cycling. I will try to recreate my fix on a different model on  my radio when I get the chance to figure out what it was.

commit 889b14df (2015-06-12): Support pan and tilt on fixed wing.  Fixes #1006

The general idea is that nothing calls pwmWriteServo except for
writeServos() and that writeServos() knows which servos it has used.