betaflight/betaflight #1023: UART3 issue Kylin 250 BetaFlight 3.0.0

NL-BlackDragon, 2016-08-17:
Hallo Boris,

Met betaflight 3.0.0 is er geen osd (mwosd 1.6.1.0) weergave meer met mijn kylin 250
(disarmed/nodata) osd krijgt geen verbinding met fc meer.
dit door een probleem met de uart 3 poort. 
De switches worden ook sinds 3.0.0-RC7 ook 2x weergegeven in de configurator.
Ik heb nog geen tijd gehad om in de code te kijken wat er is veranderd sinds v2.9.1.
Maar met 2.9.1 werkte de osd weergave nog prima.

Groet,
Stephan

borisbstyle, 2016-08-19:
Gebruik je ppm?

NL-BlackDragon, 2016-08-20:
Ik gebruik S.Bus

NL-BlackDragon, 2016-08-21:
Hallo Boris,

Met versie 3.0.0-RC8 heb ik ook geen osd weergave. Zelfde probleem als met RC6 en RC7 wat ik heb kunnen testen.
Geeft een fout op i2c. UART_1 switches worden nog steeds 2x weergegeven in de configurator (Zie mijn vorige post hier boven weergegeven).

De Kylin 250 heeft een CC3D FC. Daarbij gebruik ik de R9D receiver aangesloten via S.Bus. 
Gebruikt UART_3 voor de OSD en UART_1 voor serial_rx.

Voor nu terug naar versie 2.9.1 en alles werkt weer goed.

Hoop dat je dit probleem snel kan vinden en oplossen ;-) Uiteraard wil ik ook gebruik kunnen maken van de nieuwere versies van BetaFlight.

Mvg,
Stephan Schols

borisbstyle, 2017-01-04:
probeer 3.1 versie nogmaals.

To zover lijkt het een issue op je hardware eerder

closed 2017-01-04 by borisbstyle