betaflight/betaflight #1029: VTX Channel and Output Power Selection via UART using stick or switch commands

rdarby777, 2016-08-17:
I have a spare UART and would like to setup my Unify Pro HV using my FC. SmartAudio is a UART port so it should be possible to use it like the Core OSD would. Would be awesome to use stick commands+aux switch to setup my VTX without having to get out of my comfy chair.

Bonus: Output current band/channel/mw using SmartPort telemetry if at all possible to see the current settings on my Taranis.

See also:
cleanflight#2281
d-ronin/dRonin#1067

jflyper, 2016-08-21:
Stick commands and switch based VTX channel selection are implemented for RTC6705 in SINGULARITY, and AFAIK, in SIRINFPV targets. I'm not sure if the targets share the same RC input handling. SINRINFPV's support also includes OSD menu system integrated band selection.

It would be nice if (1) a new abstraction on VTX is created and (2) RC input handling would be united and (3) use the new abstract to drive underlying VTX hardware so that (4) new VTX hardware such as the one mentioned above could be added easily.

rdarby777, 2016-08-22:
I have the document for the SmartAudio protocol.  What is the best way to get that info in the right hands?

rdarby777, 2016-08-22:
[TBS SmartAudio Rev07.pdf](https://github.com/betaflight/betaflight/files/430302/TBS.SmartAudio.Rev07.pdf)

Here is the document...TY

NightHawk32, 2016-08-28:
I did some experimenting based on this document regarding CRC calculation and data format. But as @jflyper mentioned we first need to a new VTX abstraction and RC handling. Does anyone know who is responsible for SINGULARITY and SIRINFPV?

Regarding Smart Audio we need to introduce a new type of serial device and a way to handle bidirectional asynchronous communication. Maybe this can be done by using the task scheduler?

jflyper, 2016-08-28:
> Does anyone know who is responsible for SINGULARITY and SIRINFPV?
For SINGULARITY, it's @sblakemore.
For SIRINFPV, it's @savaga.

SINGULARITY used to use dedicated SPI bus for controlling RTC6705; newer revision may be using bit-banging (haven't checked recently). SIRINFPV bit-bangs RTC6705.

> Regarding Smart Audio we need to introduce a new type of serial device and a way to handle bidirectional asynchronous communication. Maybe this can be done by using the task scheduler?

I don't have the UNIFY vtx, but I have an impression that the SmartAudio is not an ordinary full duplex serial as it only uses 1-wire. Probably use some L2-MAC protocol for arbitration, which implies some low level driver is required.

I also remember seeing somewhere that UNIFY is also capable of being an I2C slave. Is this documented somewhere?

> But as @jflyper mentioned we first need to a new VTX abstraction and RC handling

The whole picture is actually bigger, it is linked to OSD/OLED menu system (See cleanflight/#2014).
But we can start with VTX abstraction, I guess.

NightHawk32, 2016-08-28:
Had a look at the existing targets, indeed SIRINFPV uses bit-banging, and SINGULARITY still uses hardware SPI.
SmartAudio is half duplex, with a master-slave communication. The FC sends commands or requests data, the VTX responds. So the first tests could also be done by just sending commands without even looking at the responses. It can take up to 100ms for the VTX to send a proper response.
The response is also needed to identify if SmartAudio V1 or V2 is supportetd. V2 involves some specials like pit mode ...., but some basic functions should be implemented first.

I have a Unify to do some testing on it, but I don't know anything about I2C being involved.

Yes you are right there are lots of other things to do, even MSP and configurator integration. But I think we should start things slow. So should we start this together?

jflyper, 2016-08-29:
@NightHawk32 Yeap. I just clicked for a couple of unify pros. I will probably fabricate an I2C version because UARTs are premier resource for most F3 (and F1) users; it's too regrettable to use them for something that is used for few seconds (same apply for SPI and GPIOs for bit-banging).

jflyper, 2016-09-09:
@rdarby777 Do you have any documents regarding the physical layer of the SmartAudio? I mean electrical specifications.

rdarby777, 2016-09-09:
@jflyper this is what I received from Trappy today.  Looks like he needs more specifics on what your requirements are "... not sure what you mean exactly. It's half-duplex serial data connection, all other electrical specifications I can think of are in the first page of the document. "

jflyper, 2016-09-10:
@rdarby777 
> Hardware Interface
Software Serial Port: 4800bps 1 Start bit and 2 Stop bit Voltage Level: Logic high 0.9­3.3V / Logic low 0V­0.5V

Yeah, for two wire full-duplex communication, this would be enough, but since this is an one-wire half-duplex bus, we need to know:

1. How each device drive and listen to the bus:
Open drain (open collector) or not, TX goes Hi-Z when idle or not, and such. Best information is to know how UNIFY is interfacing. Interfacing circuit such as [S.PORT example](http://eleccelerator.com/pics/x4rsb_smartport_hack/original_x4rsb_ckt_rep.png) would be great.

2. How and when each device gain access to the bus:
It looks like a "master talks, slave answers" protocol, but we don't know, for example, the min and the max between talk-answer gap. We need to know details of timing specification for correct operation and sensible error recovery.

NightHawk32, 2016-09-12:
@jflyper 
I managed to interface a Unify using just an arduino.

1. It seems like the unify uses some kind of push pull mode. The audio pins stays low between transmissions so there is no pull-up on the unify. The Unify also requires the signal to be low before and after transmission, so I transmitted front and tailing 0, just to be safe.
On the arduino I took a hardware serial port and connected rx and tx to the SmartAudio. When the transmission is complete I deactivate tx and set it to input. This is the software solution, but it could also be done this way:

[image]

2. I dont't think that the unify is really accurate, since it gets pretty hot and uses MCU internal clock. I have something in mind that the typical response time is about 100ms. But I think this can drift a little. Even the baudrate changes over time. For my Unify it starts with the specified 4800 baud and goes to 4900.

rdarby777, 2016-09-12:
SmartAudio uses single wire half duplex communication based on UART. In Idle mode both host and VTX define the SmartAudio pin as input.  If the host want to talk to the VTX, he defines the SmartAudio pin as output and start sending a command. Once the command is sent the host defines the SmartAudio pin as input and the VTX will answer according to the received command. The VTX only sends data if requested by the host. 

The VTX never talks form himself and only answers if the master sent something. More details can be fund under commands -> summary

Host to VTX
For all frames sent to the VTX the command byte needs to be shifted left by one bit and the LSB needs to be set. For example to set power the command send over the bus from the host to the VTX is 0x05 and the returned command is 0x02.
Response from VTX 
The Unify Pro response frame is usually send <100ms after a frame is successfully received from the host MCU. As the Unify Pro always will  answer for any frame with a response, this response can be used to verify if the Unify Pro successfully received the sent frame. 
The default response it the same frame as sent by the host including the updated data and an additional reserved byte at the very end of the payload. This reserved byte can be ignored.  
For the GET SETTINGS command the VTX will answer with a full set of settings and the SmartAudio version number. This frame also doesn’t include a reserved byte.

jflyper, 2016-09-12:
@NightHawk32 @rdarby777 
Thanks for the info. I'll try to match these characteristics against available UART operating modes in BF. The bps drift may present some problem, because 100/4800=2% error, which is marginal already.

@NightHawk32 
BTW, is the diode drawn correctly? And isn't it better to use schottky part for lower Vf?

jflyper, 2016-09-12:
Since the line is shared with audio input, my guess is that unify identifies digital phase and analog phase by presence of non-zero voltage on the audio line. Unify transmits whatever signal given to audio line. As soon as it detects zero on the line it disables audio transmission and see if the zero continues for at least one char period. If it doesn't, then unify will re-enable audio transmission. If the line remains zero, unify goes into digital phase, and enables receiver. The digital phase is terminated with non-zero voltage that last for one char period.

If we pull-up the audio line, then it will be passed to audio band. I'm not sure about the consequence in terms of final rf signal.

rdarby777, 2016-09-12:
Everyone knows the documentation regarding smart audio is in this thread on a pdf about four posts from the top correct?

NightHawk32, 2016-09-12:
@rdarby777 
yes, otherwise I would not know about the commands :-)

@jflyper 
Nah, used a 3.3V arduino.
I don't know what exaclty happens in the unify. If this way uses additional audio band, it should be avoided. But there is still the other option with enabling/disbaling tx, which I used on the arduino.

As long as the diode is fast enough it should not be that important which one is used exaclty.
Due to the diode TX is only able to pull the line to low, so that's why the pull-up is needed.

Machtap, 2016-09-13:
@NightHawk32 Would you be willing to share the arduino code?

jflyper, 2016-09-13:
Yeap.

jaimevisser, 2016-09-13:
Duplicates #809. As there is more actual progress here I would suggest @borisbstyle close #809.

jflyper, 2016-09-14:
There has been some progress at ShikOfTheRa/scarab-osd#315

NightHawk32, 2016-09-14:
@Machtap yes, for sure I can. It does not look very clean but I guess it's ok for testing. Will post it later.

NightHawk32, 2016-09-14:
@jflyper I've already had a look at HOTT telemetry which works in a similar way(half-duplex). It is connected as described above with the diode, but the pull-up is not needed there because the receivers seem to have one built-in.
If we want to do it this way we have to figure out what happens to the speaker when pulling the audio high and if this can be tolerated.

jflyper, 2016-09-15:
@NightHawk32 
Hardware UART driver has BIDIR operational mode (which is used for 1-wire half-duplex), which automagically turns off TX (and presumably goes Hi-Z) when the transmitter is idle. I will look if BIDIR works, but some degree of complication will be introduced by:
1. Software serial does not implement this feature.
2. External inverters (non-F3 targets) can not go Hi-Z.

For the baud rate drifting, 0x55 can be used to determine the baud rate (auto baud). I believe the start code 0xAA 0x55 wasn't chosen for no reason. A problem is that if we can do auto baud-ing on top of the current codes, and the Unify doesn't talk unless it receives a legitimate frame(?).

NightHawk32, 2016-09-15:
@jflyper BIDIR mode sounds good. That would solve the need for external components.

Also thought about doing auto baud. But you are right, the Unify only talks when a correct frame is received. An option is to combine the testing of baud rates with auto baud rate. On power up multiple baud rates are tested until the unify responds and during operation auto baud rate detection is used to compensate small drifts. Of course this won't work if we don't have regular data transmission.

jflyper, 2016-09-15:
@NightHawk32 Unify should respond to whatever received...

> As the Unify Pro will answer for any frame with a response this response can be used to verify if the Unify Pro successfully received the sent frame.

It sounds like the Unify responds to whatever heard, even for error frames.

jflyper, 2016-09-15:
Found couples of problems over at the ShikOfTheRa/scarab-osd#315.

Out of the problems, significant one is the confusion about the CRC.

jflyper, 2016-10-02:
Initial cut that works with the current BFOSD is coming up sometime this week.

It will support primitive frequency (band and channel) and transmit power control.
No pit mode or other fancy stuff yet.

sniperxnl, 2016-10-04:
Will this work on the Immersionrc Tramp?

jflyper, 2016-10-05:
@sniperxnl 
TBH, I don't know. Is protocol (physical, datalink, transport and command/response) disclosed?

sinned88, 2016-10-05:
The 4-pin JST is for the NFC chip wich is used the the "wand". The telemetry pin on the other connector looks more promising. I wrote to immersion but did not recieve an answer jet. This is what the say on the website:

"A telemetry port on the TrampHV allows tight integration with any quad equipped with a spare serial port.

Program the channel, band, and power level, along with special modes such as Pit Mode."

jflyper, 2016-10-06:
@sinned88 
Ah, the small chip on the TNR(?) module is the NFC.

So, on the other connector, I see 'T' in addition to standard A,V,G,5V and V+. Suggest single-wire protocol, not as bad as clock & data. I hope, especially for the physical layer, it's similar to SmartAudio.

jflyper, 2016-10-06:
So, anyone has any information on how to turn on the pit mode? Commands and parameters in the protocol document Rev 0.7 don't seem to work.

rdarby777, 2016-10-06:
@jflyper I can put you in touch with one of the developers via email.  Contact me at AllCarbonRc support/sales email address.

Machtap, 2016-10-06:
@jflyper 

My command functions for pitmode appear to produce the expected results (very low power output, and you can set the frequency to whatever you want before exiting pitmode.) Although they are still a little rough around the edges.

https://github.com/Machtap/SmartAudiov2

Machtap, 2016-10-06:
My functions appear to produce the expected results when sent to the unify with a mini arduino, entering pitmode renders ultra low TX power and you can change frequencies as you please before removing it from pitmode.  Not really full featured functions yet but they do appear to be working.

Hopefully this helps.
https://github.com/Machtap/SmartAudiov2

jflyper, 2016-10-06:
I could **_accidentally_** put it into the pit mode, but not at an arbitral timing.

It only goes into pit mode if power cycled during setpit-getsetting-exitpit-getsetting command loop. And once exited, it does not go in until next power cycle.

@Machtap Can you confirm enter - exit - enter sequence when you have a spare time?

jflyper, 2016-10-08:
Haven't contacted TBS tech yet, but while fiddling with mode flags, accidentally found out that it seems to be:
Set Mode 0x10 (bit 4) is vtx power off.
Read back is Get Settings mode flags 0x20 (bit 5)

DerDominik, 2016-10-11:
Great work for implementing the possibility to configure the Unify Pro HV directly from Betaflight.
I compiled your changes and I can see the Unify options in the OSD, but I can't change them.
I think I need to activate UART1 for smartaudio?! You wrote "mask value 1024" in PR1283.
What do you mean with it? Can you please explain the detailed command I need for activate it?
Many, many thanks!!

jflyper, 2016-10-12:
@DerDominik 
If you are using UART1 for connecting to Unify Pro HV, go into the CLI, and type

`serial 0 1024 115200 57600 0 115200`
`save`

mikeller, 2016-10-12:
@jflyper This should be easy to add to the 'serial config' tab of the configurator as well. Let me know if you need help with it.

msadys, 2016-10-12:
@jflyper 
For clear:
-communication is only on one line (TX)?
-response from vtx is necessary?

jflyper, 2016-10-12:
@mikeller 
Thanks! I'd let you know when the driver is mature.

@jasiucytrynowy 
Yeap. RX will be internally connected to the TX.

jasiucytrynowy, 2016-10-12:
@jflyper 
thank you for answer, what about response from vtx? I'm asking because we plannig to add smartaudio function to our vtx.

mikeller, 2016-10-12:
@jflyper: From what I can see, the only relevant config for this feature is the 1 << 10 bitmask for the serial port. As long as this does not change there will be no change required in the configurator, and we can add support for this as soon as you PR is merged.

jflyper, 2016-10-13:
@jasiucytrynowy 
smartaudio is 1-wire half-duplex protocol; command and response goes back and forth on a single wire.

So, you are creating a new vtx with smartaudio? I'd suggest you add an I2C capability onto it. IMO, using scarce uart to control vtx may be a waste of very scarce resource. In fact, I have a breadboard with PIC mcu that talks both I2C and smartaudio.

jasiucytrynowy, 2016-10-14:
@jflyper 
this is not new vtx. Now it have passthrough for SPI (only DATA na CLK signals, two wires) because our new target will have this pins (target with OSD) and smartaudio functionally will be addition for it.  I will SA on F4.

jflyper, 2016-10-21:
Okay, tbs dev finally responded, and PIT mode behavior is now completely explained.

- The PIT mode can only be activated upon power up. Pit mode bits (In-Range or Out-Range) in Set Mode command are used to preset this behavior.

- The PIT mode can be canceled and transmission commenced by Pit mode running flag in Set Mode command.

- This operation will cancel the In-Range / Out-Range presets; it will come up transmitting at next power cycle.

- To prevent this behavior and make it come up with pit mode, the "Quit Pit Mode without unsetting the pit mode flag" is used.

Other way of saying the above is:

- Pit mode bits (IR or OR) in Set Mode do not take effect immediately; it goes effective upon next power up.
- Pit mode running flag bit in Set Mode takes effect immediately.

jflyper, 2016-10-21:
I think I'm going to implement an "operational model" scheme:

1. PIT oriented mode, in which unify will always power up with pit mode. The user must explicitly commence transmission before flight.
2. FREE mode, in which unify will always power up actively transmitting.

There will be an entry in the config menu to switch between the models.

jflyper, 2016-10-22:
@jasiucytrynowy If the vtx is onboard, or at least directly controlled by FC MCU, then SmartAudio has nothing to do with it.

jasiucytrynowy, 2016-10-23:
@jflyper my vtx is standalone and have MCU.

DerDominik, 2016-10-23:
Operation Mode sounds very good. What is the range in Pit Mode?
In my mind it's perfect if you alway start in Pit Mode after powering, and with switching into armed mode also the VTX switches into normal mode

jflyper, 2016-10-31:
@DerDominik 
Automatic tx commencing is actually a use case, but some may prefer to arming without tx commencing. May be another mode switch.

closed 2016-12-20 by borisbstyle

thetobby, 2017-08-12:
Pit mode when only disarmed and broadcasting when armed would be a real nice timesaver during our local race. Any chance it could be reconsidered as an option @jflyper