betaflight/betaflight #1034: Request for Spektrum Telemetry

cdmanbg, 2016-08-18:
Hi guys, I see on version 3.0.0 rc7 supported telemetry is: 

FrSky
HoTT
SmartPort
LTM
MAVlink  - added soon

Is there a way to add Spektrum Telemetry too. As I know they released document with some specification about such connections - can be found here in RCgroups by AndyKunz

Spektrum will ship Telemetry RX very soon SPMAR6600T and AR6270T

Specification PDF can be found here: 
http://www.rcgroups.com/forums/showpost.php?p=34091390&postcount=325

cdmanbg, 2016-08-18:
Here is the PDF attached too: 
[Spektrum Telemetry Rev E.pdf](https://github.com/betaflight/betaflight/files/423935/Spektrum.Telemetry.Rev.E.pdf)

cdmanbg, 2016-08-26:
Some update here , there is new RX now from 25.Aug.2016 SPM4649T , which is Serial Telemetry capable 

product is here: 
http://www.horizonhobby.com/spektrum-dsmx-quad-race-serial-receiver-with-telemetry--p-spm4649t

And some video from Engineers which explains better
https://www.youtube.com/watch?v=PEUZMYrDJIc

cdmanbg, 2016-11-13:
Spektrum Serial telemetry docs now released by AndyKunz and Miguel Alvarez , you can find RCgroups thread here:

https://www.rcgroups.com/forums/showthread.php?t=2775731

andymac4182, 2016-11-13:
Is this a duplicate of this https://github.com/betaflight/betaflight/issues/1084#issuecomment-260172929 ?

martinbudden, 2016-11-14:
Closing, duplicate of https://github.com/betaflight/betaflight/issues/1084

closed 2016-11-14 by martinbudden